pyxem/pyxem #786: Skimage dependency update

CSSFrancis, 2021-12-16:
**Describe the bug**
With the release of Skimage 0.19 there are a couple of bugs that have cropped up.

Changes:

- skimage.draw.circle --> skimage.draw.disk
- skimage.registration.phase_cross_correlation has some small changes that appear to be causing some tests to fail. 
- skimage.morphology.watershed --> skimage.segementation.watershed.

I've worked through most of these bugs but there appears to be some issue with the phase_cross_correlation test that I am still working on.

hakonanes, 2022-01-12:
Note that `skimage.draw.disk` has been available since 0.17 at the latest, which is the minimal supported version in pyxem, so changing to this function everywhere, and changing parameters accordingly, should be safe.

pc494, 2022-02-01:
@CSSFrancis I was thinking about having a stab at the last item on the list this weekend, can you confirm that you've not fixed it somewhere else so I can avoid duplicating work?

CSSFrancis, 2022-02-01:
@pc494 you can take a stab at it just to get the tests passing, otherwise I can do it as well.  There are some larger problems with the `phase_cross_correlation` function that I have started to fix here:  https://github.com/scikit-image/scikit-image/pull/5573 in order to get #750 working.

Long story short is that this is kind of stuck in limbo and it might be best to just patch this before fixing everything.

pc494, 2022-02-02:
Okay, I think my main aim was to get the test suite passing, even if it costs us a little bit of coverage in the short term. Will have a proper look at the weekend.

magnunor, 2022-02-03:
Some of the tests are also failing due to changes in the peak finding, for example `blob_dog`. Not sure exactly why, I'll have a look at it. Some of tests which are failing:

```python
FAILED pyxem/tests/utils/test_dask_tools.py::TestPeakFindDog::test_single_frame_one_peak[112-32]
FAILED pyxem/tests/utils/test_dask_tools.py::TestPeakFindDog::test_single_frame_one_peak[170-92]
FAILED pyxem/tests/utils/test_dask_tools.py::TestPeakFindDog::test_single_frame_one_peak[54-76]
FAILED pyxem/tests/utils/test_dask_tools.py::TestPeakFindDog::test_single_frame_one_peak[10-15]
FAILED pyxem/tests/utils/test_dask_tools.py::TestPeakFindDog::test_single_frame_multiple_peak
FAILED pyxem/tests/utils/test_dask_tools.py::TestPeakFindDog::test_single_frame_threshold
```

magnunor, 2022-02-03:
With regards to `blob_dog`, it seems to be related to the `sigma_ratio` parameter. Not sure if it is a bug fix in `skimage` or a regression.

---------

To test this:

```python
import numpy as np
from skimage.feature import blob_dog
image = np.zeros(shape=(200, 100), dtype=np.float64)
peak_list = [[120, 76], [23, 54], [32, 78], [10, 15]]
for x, y in peak_list:
    image[x, y] = 654
min_sigma, max_sigma, sigma_ratio = 2, 5, 5
threshold, overlap = 0.01, 1
normalize_value = None
if normalize_value is None:
    normalize_value = np.max(image)
peaks = blob_dog(
    image / normalize_value,
    min_sigma=min_sigma,
    max_sigma=max_sigma,
    sigma_ratio=sigma_ratio,
    threshold=threshold,
    overlap=overlap,
    exclude_border=False,
)
print(peaks)
```

- `skimage==0.18.3`: `[[120.  76.   2.], [ 32.  78.   2.], [ 23.  54.   2.], [ 10.  15.   2.]]`
- `skimage==0.19.0`: `[]`
- `skimage==0.19.0`, with `sigma_ratio=1.6`: `[[120.  76.   2.], [ 32.  78.   2.], [ 23.  54.   2.], [ 10.  15.   2.]]`

magnunor, 2022-02-05:
The `blob_dog` changes are due to this pull request in `scikit-image`: https://github.com/scikit-image/scikit-image/pull/5503

I'll update the pyxem functions to take this into account.

magnunor, 2022-02-05:
> The blob_dog changes are due to this pull request in scikit-image: https://github.com/scikit-image/scikit-image/pull/5503
>
> I'll update the pyxem functions to take this into account.

Fixed in https://github.com/pyxem/pyxem/pull/798

closed 2024-01-26 by pc494