pyxem/pyxem #788: Version 0.14.0

pc494, 2022-01-06:
Happy New Year to all; in exciting news https://github.com/hyperspy/hyperspy/pull/2703 is now merged, so I think we should start thinking about getting the code into a release state for when that comes out. I'll write a list up tomorrow, but in the meantime if people have ideas they could drop in the comments that would up a lot.

magnunor, 2022-01-06:
I very much agree! Should we skip the 0.13.x release, and go straight for 0.14.0?

pc494, 2022-01-06:
I think so, there is some new functionality floating around + some deprecations I would like to get in that make a minor release make sense (at least to me) + it will be over a year since 0.13.0

magnunor, 2022-01-06:
Sounds good! I'll start on a pull request which makes pyxem work with the hyperspy `RELEASE_next_minor` (ergo, the 1.7.0 version)

magnunor, 2022-01-20:
There are a number of tests from my "old" pixstem code which is currently failing, I'll go through those and sort them out. Probably during the weekend.

Most of them seems related to various skimage functions like `EllipseModel` and peak finding.

pc494, 2022-01-20:
Pruning (or deprecating) of anything that can go is also encouraged.

Not sure if I've said this already, but there is no huge rush on this.

magnunor, 2022-03-13:
With the further `map` changes sorted out by @CSSFrancis in https://github.com/hyperspy/hyperspy/pull/2878, I think the remaining parts includes fixing the various broken functionalities in https://github.com/pyxem/pyxem/tree/hyperspy_release_next_minor

I sorted out some of these in https://github.com/pyxem/pyxem/pull/789, but there were a number of issues related to the handling of ragged arrays.

@CSSFrancis started on this in the https://github.com/hyperspy/hyperspy/pull/2877 pull request, where the work on `blockwise` was separated out into a new pull request (https://github.com/hyperspy/hyperspy/pull/2877).

@CSSFrancis: any plans to continue with the "ragged array" part of https://github.com/hyperspy/hyperspy/pull/2877? If yes, we can wait with those specific "bugs" in pyxem until this is resolved. If not, we can sort out the issues in https://github.com/pyxem/pyxem/tree/hyperspy_release_next_minor in pyxem with some workarounds.

magnunor, 2022-03-15:
@CSSFrancis, sounds good! I got through some of the failing tests in https://github.com/pyxem/pyxem/pull/789, so see if those solutions can be copied.

I had planned "copying" the fixes from https://github.com/pyxem/pyxem/pull/789 to https://github.com/pyxem/pyxem/tree/hyperspy_release_next_minor, with some extra care with regards to the recent changes in HyperSpy, but I haven't had the time yet.

CSSFrancis, 2022-03-15:
So far we have the following (Grouped) failing tests

- [ ] 0D diffraction Vector Fixes: #811 
- [ ] Pearson Correlation Bug Fixes: #809 
- [ ] Subpixelrefinement Generator bug fixes @magnunor can you submit your fix for this one when you have the chance.
- [ ]  Index generator --> dict changes
- [ ] `map` function with an iterating signal that is a function rather than an array (@CSSFrancis to submit fix to hyperspy)
- [ ] Index Generation fails (`test_indexation_generator.test_vector_indexation_generator_index_vectors`)  This is related to DiffractionVectorLibrary.structures not being properly initialized in that test. This might be something that @hakonanes might have a better idea about as I don't really know much about diffsims.  That being said that test suite is kind of confusing so I wouldn't be opposed to rewriting it.  Long term this might be something I might plan to fix with an overhaul of the diffraction vectors class.
- [ ] Radial Integration map functions @magnunor is the plan still to remove these in 0.14?  I think that is probably a decent idea.  That fixes those issues and cleans up some of the duplication.

magnunor, 2022-03-15:
> Subpixelrefinement Generator bug fixes @magnunor can you submit your fix for this one when you have the chance. I got a little confused with the fix for this one. (I think the previous method wasn't doing what it was supposed to anyways)

Indeed! I think I got this all sorted in my old pull request, I'll look at it soonish.

-------------

> Radial Integration map functions @magnunor is the plan still to remove these in 0.14? I think that is probably a decent idea. That fixes those issues and cleans up some of the duplication. I think the errors there are related to directly calling _map_iterate which is now not recommended. Let me know if you want me to do this sometime this week.

The failing tests should be fixed now (I cherry-picked the commits which fixed this). With regards to removing `radial_integration` I agree!

-------------

There also seems to be a bug affecting just the windows tests: https://github.com/pyxem/pyxem/runs/5560089211?check_suite_focus=true#step:8:60, seems like it is related to `dask` threading?

I haven't had time to examine it properly, but it does not seem to happen to the `master` branch, where I just now ran the windows 3.8 workflow, which passed: https://github.com/pyxem/pyxem/runs/5560255963?check_suite_focus=true

magnunor, 2022-03-20:
The subpixelrefinement generator and the integration generator issues is now fixed!

That leaves:
- Index Generation
- Pearson Correlation
- The windows(?) bug, for example: https://github.com/pyxem/pyxem/runs/5617641913?check_suite_focus=true#step:8:189

pc494, 2022-03-21:
Okay these are probably me problems, I should be able to have a look this week :)

pc494, 2022-03-26:
> The indexation generator is failing in two tests:
> 
>     * `pyxem/pyxem/tests/signals/test_indexation_results.py:61`: `AttributeError: 'dict' object has no attribute 'get_library_entry'`
> 
>     * `pyxem/pyxem/tests/generators/test_indexation_generator.py:237`: `AttributeError: 'dict' object has no attribute 'structures'`
> 
> 
> The windows bug seems to fixed in #809, thus the only thing missing is the `indexation` issues.

I think both of these relate to the old version of template matching, and so I would be in favour of dropping the tests as we move towards using the new improvement version in all future applications.

CSSFrancis, 2022-03-27:
> I think both of these relate to the old version of template matching, and so I would be in favour of dropping the tests as we move towards using the new improvement version in all future applications.

I'm okay with this.  It would be good to take a look after 0.14 is released at what our goals are for pyxem are over the next year or so. Primarily from the standpoint of what stays and what goes.  

Is that plan still to try to push for a 1.0 release down the road? I know that is probably a fair bit of work so maybe something like creating a road map of releases might be a good idea.

From at least my development goals I have the following development goals at least before anything like a stable 1.0 release:

- Overhaul Diffraction Vectors class and Matching
 
- Overhaul Calibration generator and astigmatism correction

- Depreciate and remove some generators in favor of better documentation. (i.e. remove virtual dark field generator in favor of using a hyperspy roi)

Dropping tests is probably fine in the short run, but we should maybe have a defined way forward if things are broken and need fixing.

hakonanes, 2022-03-27:
> remove virtual dark field generator

In [kikuchipy](https://kikuchipy.org/en/stable/user_guide/virtual_backscatter_electron_imaging.html), I find having a virtual backscatter electron imaging (VBSE) generator (EBSD equivalent of VDF/VBF generator) which provides convenience methods for common uses of HyperSpy ROIs (often multiple ROIs) very nice. Of course one shouldn't hide the flexibility of ROIs from users, but providing shortcuts doesn't hurt.

> Depreciate and remove some generators

The sooner people are notified that some functionality will be removed or replaced, the better. So if we agree that some functionality will be removed, and an alternative exist, marking something as deprecated doesn't cost anything.

CSSFrancis, 2022-03-28:
With #817 we should be back to all of the tests passing.  The 2 remaining tests were just ignored which per @pc494 is probably good for now.  

That should be rewritten in the next couple of months in which case I will replace those tests.

magnunor, 2022-03-29:
I think the last thing which needs to be sorted out, is the changes due to https://github.com/hyperspy/hyperspy/pull/2830: several of the `axes_manager` attributes will now be protected, and can't be changed directly.

We can't see these issues in the unit tests, as it hasn't been merged yet. However, it should be fairly straight forward to fix these.

hakonanes, 2022-03-29:
> We can't see these issues in the unit tests

But you get an idea by inspecting the failing tests when the [extension/integration tests](https://github.com/hyperspy/hyperspy/runs/5650937307?check_suite_focus=true) were run in that PR seven days ago.

CSSFrancis, 2022-04-07:
It seems like the next thing we should look at before this release is the demos.  I'll start going through them with Hyperspy version 1.7 and 0.14 this weekend and see if everything works fairly smoothly or needs some additional work.

din14970, 2022-04-07:
@CSSFrancis just a heads up, for sure the template matching demo will be broken as I messed with some of the arguments in the functions. I will have to do a clean-up.

magnunor, 2022-04-24:
There are currently 3 tests failing in the [hyperspy_release_next_minor](https://github.com/pyxem/pyxem/tree/hyperspy_release_next_minor) branch. Not 100% sure what caused them to break.

```python
 FAILED pyxem/tests/generators/test_subpixelrefinement_generator.py::Test_init_xfails::test_out_of_range_vectors_DiffractionVectors
FAILED pyxem/tests/signals/test_indexation_results.py::test_single_vector_get_crystallographic_map
FAILED pyxem/tests/signals/test_indexation_results.py::test_double_vector_get_crystallographic_map
```
-------------------

<details><summary>Full test log, click to expand</summary>
<p>

```python
________ Test_init_xfails.test_out_of_range_vectors_DiffractionVectors _________

self = <pyxem.tests.generators.test_subpixelrefinement_generator.Test_init_xfails object at 0x7fbeca352160>

    def test_out_of_range_vectors_DiffractionVectors(self):
        """Test that putting vectors that lie outside of the
        diffraction patterns raises a ValueError"""
        vectors = DiffractionVectors(np.array([[1, -100]]))
        dp = ElectronDiffraction2D(np.ones((20, 20)))
    
        with pytest.raises(
            ValueError,
            match="Some of your vectors do not lie within your diffraction pattern",
        ):
>           _ = SubpixelrefinementGenerator(dp, vectors)

/home/runner/work/pyxem/pyxem/pyxem/tests/generators/test_subpixelrefinement_generator.py:84: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <pyxem.generators.subpixelrefinement_generator.SubpixelrefinementGenerator object at 0x7fbec33f0790>
dp = <ElectronDiffraction2D, title: , dimensions: (|20, 20)>
vectors = <DiffractionVectors, title: , dimensions: (2, 1|)>

    def __init__(self, dp, vectors):
        self.dp = dp
        self.vectors_init = vectors
        self.last_method = None
        sig_ax = dp.axes_manager.signal_axes
        self.calibration = [sig_ax[0].scale, sig_ax[1].scale]
        self.center = [sig_ax[0].size / 2, sig_ax[1].size / 2]
    
>       self.vector_pixels = _get_pixel_vectors(
            dp, vectors, calibration=self.calibration, center=self.center
        )

/home/runner/work/pyxem/pyxem/pyxem/generators/subpixelrefinement_generator.py:274: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

dp = <ElectronDiffraction2D, title: , dimensions: (|20, 20)>
vectors = <DiffractionVectors, title: , dimensions: (2, 1|)>
calibration = [1.0, 1.0], center = [10.0, 10.0]

    def _get_pixel_vectors(dp, vectors, calibration, center):
        """Get the pixel coordinates for the given diffraction
        pattern and vectors.
    
        Parameters
        ----------
        dp: :obj:`pyxem.signals.ElectronDiffraction2D`
            Instance of ElectronDiffraction2D
        vectors : :obj:`pyxem.signals.diffraction_vectors.DiffractionVectors`
            List of diffraction vectors
        calibration : [float, float]
            Calibration values
        center : float, float
            Image origin in pixel coordinates
    
[... 195 lines cut ...]
        inputs when the selection is smaller than half of the original array. This
        avoids excess memory usage when extracting a small portion from a large array.
        For more information, see
        https://numpy.org/doc/stable/reference/arrays.indexing.html#basic-slicing-and-indexing.
    
        Parameters
        ----------
        obj: ndarray, string, tuple, list
            Object to get item from.
        index: int, list[int], slice()
            Desired selection to extract from obj.
    
        Returns
        -------
        Selection obj[index]
    
        """
>       result = obj[index]
E       TypeError: tuple indices must be integers or slices, not tuple

/opt/hostedtoolcache/Python/3.8.12/x64/lib/python3.8/site-packages/dask/array/chunk.py:417: TypeError
```

</p>
</details>

CSSFrancis, 2022-04-26:
@pc494  Can we just skip these tests? What do we still need for the 0.14 release? 

I've ran through a couple of the notebooks and everything seems to be working for the most part.  I have some additional information I've been meaning to add there that I'll do later.

As far as the status of the outstanding PR's it would be good to merge:

#829 
#827 
#826 
#823

magnunor, 2022-04-27:
Sounds good! I'm going through those pull requests now.

We also need to merge the `hyperspy_release_next_minor` into the `master` branch. I can do that.

I suggest we wrap up the outstanding pull requests, merge `hyperspy_release_next_minor`, then check if everything is working, lastly we do a release.

I think @pc494 needs to do the release itself?

pc494, 2022-04-27:
Good work everyone, I think it may well be true that I need to perform this release. I'll get to work on my outstanding bits (which appear to be a fair amount of the problems)

magnunor, 2022-04-27:
I made a pull request for merging the `hyperspy_release_next_minor` branch into the `master` branch: https://github.com/pyxem/pyxem/pull/832

Given that no further tests fail, that pull request should be ready for merging.

Other than that, there is the indexation pull request (https://github.com/pyxem/pyxem/pull/823), and fixing any currently failing tests.

magnunor, 2022-04-27:
One thing to remember: increasing the HyperSpy version in the conda-forge version of pyxem

CSSFrancis, 2022-04-27:
It might be a good idea to manually add #824 as well before releasing. I can do that maybe later tonight once everything is merged to the master branch.

magnunor, 2022-04-27:
In addition to the 3 unit tests already failing, there are 2 new ones:

```python
FAILED pyxem/tests/signals/test_beam_shift.py::TestFullDirectBeamCentering::test_mask
FAILED pyxem/tests/signals/test_beam_shift.py::TestFullDirectBeamCentering::test_mask_lazy
```
We should probably have a look at those after everything is merged into the `master` branch.

-----------------

@CSSFrancis, feel free to merge https://github.com/pyxem/pyxem/pull/832 after https://github.com/pyxem/pyxem/pull/833 is ready.

------------------

Should we aim for a release tomorrow? So that everything is ready for a HyperSpy bundle release.

pc494, 2022-04-28:
Once #823 passes tests I'll merge that, then I'll do #824 and then I'll package a release if that works for everyone?

magnunor, 2022-04-28:
Sounds good! I'm trying to figure out why the `test_beam_shift.py` tests are failing, but currently they're passing locally on my computer.

magnunor, 2022-04-28:
> Sounds good! I'm trying to figure out why the test_beam_shift.py tests are failing, but currently they're passing locally on my computer.

This is resolved now! Still 3 tests failing, but not sure how to fix them (or if we need to fix them...).

pc494, 2022-04-28:
Will deal with the tests and package the release tomorrow :)

pc494, 2022-04-29:
this is now underway, closing.

magnunor, 2022-04-30:
@pc494, status on the conda-forge version?

pc494, 2022-04-30:
It's failing on a comparison within the testing suite, I've added you as a maintainer in the PR while I was there.

pc494, 2022-04-30:
conda forge is now done (it was a dependencies issue)!