pyxem/pyxem #806: Switch master branch to HyperSpy Rnm for automatic unit tests?

magnunor, 2022-02-27:
With the release of HyperSpy 1.7 (https://github.com/hyperspy/hyperspy/issues/2845) getting closer, we should maybe switch over the `master` branch to use the HyperSpy's `RELEASE_next_minor` for the automatic unit tests.

This so we more easily can see what works, and what doesn't work.

This is necessitated by changes in how the `signal.map` functionality works in HyperSpy, which breaks a number of pyxem's functions.

hakonanes, 2022-02-28:
Either that or add a new branch installing the `RELEASE_next_minor` version of HyperSpy in CI.

magnunor, 2022-03-01:
I added a new branch with this directly in pyxem's repository: https://github.com/pyxem/pyxem/tree/hyperspy_release_next_minor

For the various bug-fixes related to the new hyperspy release, we should make pull requests to that branch.