pyxem/pyxem #813: Make codebase conform to black-style checker

magnunor, 2022-03-15:
`pyxem` follows the black-style formatter, however currently 55 files has some kind of style error. Using `python3 -m black --diff --check .` in the pyxem folder: `55 files would be reformatted`.

So I suggest two things:
- Right after the next pyxem release, we should use the automatic black formatter to fix the style in all the files, for the development branch.
- Add an automatic style checker to the automatic tests, which can be something similar as in Atomap: https://gitlab.com/atomap/atomap/-/blob/master/.gitlab-ci.yml#L10

hakonanes, 2022-03-15:
We do this on every commit to a branch or PR in its own action: https://github.com/pyxem/kikuchipy/blob/853371528be59cb502d438f6647df1f614971603/.github/workflows/build.yml#L6-L11

magnunor, 2022-06-21:
I'm tempted to do this soon, as there aren't that many big outstanding pull request now.

pc494, 2022-08-13:
#865 closes this.

closed 2022-08-13 by pc494