pyxem/pyxem #818: Pyxem orientation mapping paper available online

din14970, 2022-03-29:
Friends, the Pyxem orientation mapping paper is now available online https://www.sciencedirect.com/science/article/pii/S0304399122000511. MPIE intends to pay for the open access fee. I have not yet received proofs, but perhaps some of the citation details like DOI can already be added to the readme or documentation.

hakonanes, 2022-03-30:
That's great! (I am a co-author.) Perhaps a second sentence after "If analysis using pyxem forms a part of published work please cite the DOI at the top of this page." in the README stating that we would appreciate an additional citation of the paper if the orientation mapping is used in published work? In the documentation, a similar note on the [Citing pyxem](https://pyxem.readthedocs.io/en/latest/cite.html) page should be added.

hakonanes, 2022-04-05:
The bibtex entry is now visible in the README and on Read the Docs [here](https://pyxem.readthedocs.io/en/latest/cite.html).

closed 2022-04-05 by hakonanes