pyxem/pyxem #841: test failure with scikit-learn 1.1.0rc1

ericpre, 2022-05-03:
from https://github.com/hyperspy/hyperspy-extensions-list/pull/22
```python
_________________ TestLearningSegment.test_learning_ncc_matrix _________________

self = <pyxem.tests.signals.test_segments.TestLearningSegment object at 0x7f5c2e79f3d0>
learning_segment = <pyxem.signals.segments.LearningSegment object at 0x7f5c17c1bfa0>

    def test_learning_ncc_matrix(self, learning_segment):
        ncc = learning_segment.get_ncc_matrix()
        ans = np.array(
            [
                [
                    [1.0, -0.26413543, -0.50636968, 0.61237256, 0.0],
                    [-0.26413543, 1.0, -0.40125028, -0.43133[109](https://github.com/hyperspy/hyperspy-extensions-list/runs/6272585445?check_suite_focus=true#step:15:109), 0.0],
                    [-0.50636968, -0.40125028, 1.0, -0.31008684, 0.0],
                    [0.61237256, -0.43133109, -0.31008684, 1.0, 0.0],
                    [0.0, 0.0, 0.0, 0.0, 1.0],
                ],
                [
                    [1.0, -0.0588285, -0.07313363, -0.04087783, 0.0],
                    [-0.0588285, 1.0, -0.07312725, -0.04099601, 0.0],
                    [-0.07313363, -0.07312725, 1.0, -0.05096472, 0.0],
                    [-0.04087783, -0.04099601, -0.05096472, 1.0, 0.0],
                    [0.0, 0.0, 0.0, 0.0, 1.0],
                ],
            ]
        )
>       np.testing.assert_almost_equal(ncc.data, ans)
E       AssertionError: 
E       Arrays are not almost equal to 7 decimals
E       
E       Mismatched elements: 12 / 50 (24%)
E       Max absolute difference: 0.00049256
E       Max relative difference: 0.01204947
E        x: array([[[ 1.0000000e+00, -2.6413860e-01, -5.0636968e-01,  6.1237492e-01,
E                 0.0000000e+00],
E               [-2.6413860e-01,  1.0000000e+00, -4.0125023e-01, -4.3133[112](https://github.com/hyperspy/hyperspy-extensions-list/runs/6272585445?check_suite_focus=true#step:15:112)e-01,...
E        y: array([[[ 1.       , -0.2641354, -0.5063697,  0.6[123](https://github.com/hyperspy/hyperspy-extensions-list/runs/6272585445?check_suite_focus=true#step:15:123)726,  0.       ],
E               [-0.2641354,  1.       , -0.40[125](https://github.com/hyperspy/hyperspy-extensions-list/runs/6272585445?check_suite_focus=true#step:15:125)03, -0.43[133](https://github.com/hyperspy/hyperspy-extensions-list/runs/6272585445?check_suite_focus=true#step:15:133)11,  0.       ],
E               [-0.5063697, -0.4012503,  1.       , -0.3100868,  0.       ],...

/usr/share/miniconda3/lib/python3.9/site-packages/pyxem/tests/signals/test_segments.py:94: AssertionError
---------------------------- Captured stdout setup -----------------------------
Decomposition info:
  normalize_poissonian_noise=False
  algorithm=NMF
  output_dimension=5
  centre=None
scikit-learn estimator:
NMF(n_components=5)
```

hakonanes, 2022-05-03:
Seems to me like adding a tolerance to 1e-3 should fix this.

pc494, 2023-01-19:
#873 dealt with this.

closed 2023-01-19 by pc494