pyxem/pyxem #842: 0.13.x branch commits in master branch?

magnunor, 2022-05-07:
There are currently `16` commits which are in the `0.13.x` branch, but which are not in the current `master` branch: https://github.com/pyxem/pyxem/compare/0.13.x

Many of them are merge commits, but some of them seem to have code changes.

However, I'm not sure if there are things which were merged into the `master` branch in some other manner, or if we're currently missing some code which should be merge into the `master` branch.

magnunor, 2022-05-11:
@pc494, you might know something about this.

pc494, 2022-05-12:
Sorry, I should have written back originally, it's in my queue for post-workshop to go through this manually and tidy it up

pc494, 2022-06-02:
Okay, I really made a mess of that. The only unmerged code is detailed here: https://github.com/pyxem/pyxem/pull/850#issuecomment-1144702919 - I'll deal with this.

PR #855 (2022-06-15): Release Version 0.14.2
---
name: Release Version 0.14.2
about: Includes one bonus patch change (which will close #842)

---

**Checklist**
- [x] Final patch that @pc494 wanted via cherry picking
- [x] diffsims bumped to most modern version for future simplicity
- [x] Update `.zenodo.json`
- [x] Update `release_info` to version number and correct order of contributors

@hakonanes unless you think a `rc` is necessary I think this is good to go. Your careful eye going over it would be much appreciated.
closed 2022-06-15 by pc494