pyxem/pyxem #853: Issue with in-plane orientations from index_dataset_with_template_rotation function in indexation_utils

torit493, 2022-06-14:
**Describe the bug**
When running the index_dataset_with_template_rotation the returned orientations are sometimes larger than 360 for the in plane orientation. And the orientations don't seem to match the real rotation.

**To Reproduce**
The attached code (zip at bottom of page) reproduce the problem. It loads a small 2x2 dataset with DPs from Al and a thetaprime precipitate. Simulating diffraction patterns for cubic rotations and tetragonal rotations and loading a cif file for Al and thetaprime precipitate is used to create a diffraction library with two phases. The index_dataset_with_template_rotation is then used on the dataset to template match it. When looking at the in-plane rotations they are larger than 360 degrees and do not correctly rotate the simulations to match the DPs. It looks like this:

# Importing packages
    import numpy as np
    import matplotlib.pyplot as plt
    import hyperspy.api as hs
    import pyxem as pxm
    
    import diffpy
    from diffsims.libraries.structure_library import StructureLibrary
    from diffsims.generators.diffraction_generator import DiffractionGenerator
    from diffsims.generators.library_generator import DiffractionLibraryGenerator
    from diffsims.generators.rotation_list_generators import get_beam_directions_grid

    from pyxem.utils import indexation_utils as iutls
    from pyxem.utils.plotting_utils import plot_template_over_pattern

# Some setup parameters
    s = 0.02                          # max excitation error in simulations
    minimum_intensity = 1E-20         # min intensity included in simulations
    diffraction_calibration = 0.01904 # Å-1
    reciprocal_radius = 1.73          # Å-1
    half_shape = (64, 64)             # pixels

# Loading file 
    file = 'Al_ThP_DPs.hdf5'
    experimental_data = hs.load(file)
    experimental_data.set_signal_type('electron_diffraction')
[image]
[image]

# Importing structure
    structure_Al     = diffpy.structure.loadStructure(r'Al.cif')
    structure_Theta  = diffpy.structure.loadStructure(r'thetaprime.cif')

# Getting rotations
    resolution = 0.5 # maximum angle in degrees between nearest templates. Pretty rough grid for speed.
    cubic_grid = get_beam_directions_grid("cubic", resolution, mesh="spherified_cube_edge") #Al
    tetra_grid = get_beam_directions_grid('tetragonal', resolution, mesh='spherified_cube_edge') #ThP
  
    rot_grids = [cubic_grid, tetra_grid]
    
    phases = ['Al', 'ThetaPrime001']
    
    structures = [structure_Al, structure_Theta]
  
  # Collect all in a structure library:
    library_phases = StructureLibrary(phases, structures, rot_grids)
    
    diff_gen = DiffractionGenerator(accelerating_voltage= 200,
                                    precession_angle    = 0,
                                    shape_factor_model  = "linear",
                                    minimum_intensity   = minimum_intensity, 
                                    )
    lib_gen = DiffractionLibraryGenerator(diff_gen)

    diff_lib = lib_gen.get_diffraction_library(library_phases,
                                               calibration         = diffraction_calibration,
                                               reciprocal_radius   = reciprocal_radius,
                                               half_shape          = half_shape,
                                               with_direct_beam    = False,
                                               max_excitation_error= s)

# An intensity transform function
    shift = 0.001
    
    def log_shift(raw):
        ''''
        Parameters
        -------
        raw : np.array((128, 128)). The raw data.
        shift : float. Introduces a shift for the log. To account for pixels with 0 value.
        
        Returns
        -------
        log_shift : np.array((128,128)). The log of the raw data.
        '''
        
        log_shift = np.log10(raw+shift) - np.log10(shift)
        return log_shift

# Template matching
    delta_r     = 1                     # res in inkreasing k value 
    delta_theta = 1                     # res in rotation, azimuthal angle
    max_r = experimental_data.axes_manager[2].size//2  # Max k to check IN PIXELS, Default (None) is from center of DP to corner.
    intensity_transform_function = log_shift # Function to apply to both image and template intensities on an element by element basis prior to comparison. Note that the function is performed on the CPU.
    find_direct_beam     = False        # 
    direct_beam_position = None         # 
    normalize_image      = True         # Normalize the images in the correlation coefficient calculation
    normalize_templates  = True         # Normalize the templates in the correlation coefficient calculation
    frac_keep = 1                       # 
    n_keep    = None                    # 
    n_best    = 5                       # keeping 5 best matches

    result, phasedict = iutls.index_dataset_with_template_rotation(experimental_data,
                                                                    diff_lib,
                                                                    n_best                       = n_best,
                                                                    frac_keep                    = frac_keep,
                                                                    n_keep                       = n_keep,
                                                                    delta_r                      = delta_r,
                                                                    delta_theta                  = delta_theta,
                                                                    max_r                        = max_r,
                                                                    intensity_transform_function = intensity_transform_function,
                                                                    normalize_images             = normalize_image,
                                                                    normalize_templates          = normalize_templates,
                                                                    )

    phase_map = hs.signals.BaseSignal(result['phase_index'][:,:,0])

# Looking at the solution
    px = 1
    py = 0
    n_sol = 0
    
    # Query the necessary info from the solution
    solution       = result["orientation"]
    in_plane_angle = solution[py, px, n_sol, 0] # Orientations, the first angle is the in plane angle
    sim_sol_index  = result["template_index"][py, px, n_sol]   # Collects template index at (py,px) in solution map n
    mirrored_sol   = result["mirrored_template"][py, px, n_sol] # mirrored template represents whether the original template best fits (False) or the mirror image (True)
    print('Orientation of (',px,',', py,')','is',solution[py, px, n_sol])
    print('in_plane_angle =', in_plane_angle)
    
    # Query the appropriate template:
    found_phase = phases[result['phase_index'][py, px, n_sol]]
    print(found_phase)
    
    used_sim = diff_lib[found_phase]["simulations"][sim_sol_index]  # Collects the simulation fitted from template match
    print('Orientation before matching procedure:', diff_lib[found_phase]["orientations"][sim_sol_index])
    
    fig, ax = plt.subplots(ncols=2, figsize=(8, 4))
    ax[0].imshow(phase_map)
    ax[0].scatter([px], [py], c="red", s=50)
    
    # plotting the diffraction pattern and template
    plot_template_over_pattern(experimental_data.inav[px,py].data,   # DP at px, py
                                     used_sim,                       # simulated pattern found to fit best for DP at px, py
                                     ax=ax[1],                       # Optional, to get DP in same fig as correlation map
                                     in_plane_angle = in_plane_angle,#
                                     coordinate_system = "cartesian", 
                                     size_factor = 10,               # This only changes size of plotted X for DP spot marking, not real change
                                     vmax=0.01,
                                     mirrored_template=mirrored_sol,
                                     find_direct_beam=False,
                                     cmap = "inferno",
                                     marker_color = "green"
                                    )
    for i in ax:
        i.axis("off")

[inPlaneOrientationProblem.zip](https://github.com/pyxem/pyxem/files/8900291/inPlaneOrientationProblem.zip)

**Screenshots**
[image]
As can be seen from the image the found in plane rotation does not map the simulation correctly to the experimental DP.

hakonanes, 2022-06-14:
Thank you for the bug report, @torit493.

You've attached CIF files and the notebook, but could you also provide one of the troublesome patterns in a format easily read back into Python, so that others can debug your issue? I myself don't have time at the moment. Let's hope someone else has.

> the returned orientations are sometimes larger than 360 for the in plane orientation

As discussed via email (with @tinabe, @emichr and @soupmongoose), an in-plane angle greater than 360 degrees is not an issue *in general* (gets sorted when converting to unit quaternions with orix), but it might be a consequence of the bug... I don't know.

torit493, 2022-06-15:
I forgot to include the hdf5, here is a new updated zip:
[inPlaneOrientationProblem.zip](https://github.com/pyxem/pyxem/files/8906797/inPlaneOrientationProblem.zip)
 
I also updated the one already in the issue

soupmongoose, 2022-06-19:
So, for now I will just add some observations, I will try and look more into the actual code in the future.  From what I can tell the in-plane angle is being added between the separate phases.  Example: If the pattern (0,0,90) is found to have and in-plane angle of 60 in phase_a and 60 in phase_b, the resultant in-plane reported from the matching is 120, or the euler is (120,0,90).  However it is only the in-plane angle that the function returns, it seems to use the correct in-plane when determining the answer.  This is also why sometimes the in-plane is over 360 when it shouldn't be.

hakonanes, 2022-06-20:
> So, for now I will just add some observations, I will try and look more into the actual code in the future. From what I can tell

Thanks @soupmongoose, if you can confirm this observation by pointing to a specific piece of code that would be good! The way you describe it, sounds very much like a bug.

soupmongoose, 2022-06-20:
So if we create a diffraction library with multiple phases:

```
rot_grids = [cubic_grid, cubic_grid, cubic_grid]

phases = ['Al1', 'Al2', 'Al3']

structures = [structure_Al, structure_Al, structure_Al]

# Collect all in a structure library:
library_phases = StructureLibrary(phases, structures, rot_grids)
```

Then match with all the phases we get the first three best matches as:
```
 [180.          -1.         -90.        ]
 [180.          -1.         -90.        ]
 [180.          -1.         -90.        ]
```
with the three "different" phases as their phase ids.

If we match with only one phase however the best match is [ 60.          -1.         -90.        ].  Same pattern with only the inplane angle changed.  Similar if you do 2 phases, you get [ 120.          -1.         -90.        ].  It does appear like the matching is working correctly but what gets returned in the results dict is wrong.

[Problematic orientations_forAlv2_addedphase.zip](https://github.com/pyxem/pyxem/files/8939096/Problematic.orientations_forAlv2_addedphase.zip)

hakonanes, 2022-06-20:
I don't have time to figure it out now, but it might be that the bug is somewhere in the creation of the results dictionary here:

https://github.com/pyxem/pyxem/blob/0761e5cb2b7756942d320b5ec140a7fb64cf92be/pyxem/utils/indexation_utils.py#L1600-L1630

pc494, 2023-01-16:
Can't see this being actioned, closing.

hakonanes, 2023-01-20:
> Can't see this being actioned, closing.

You might be right. I believe this is still an issue for @torit493 and @tinabe, though. It might be safest to leave it open until someone finds a solution or a way around the issue. I don't see a problem releasing a v1.0 while there still are open issues.

closed 2023-03-19 by CSSFrancis