pyxem/pyxem #854: Patch release 0.14.2

hakonanes, 2022-06-15:
Sorry for the short notice, but would it be possible to make a patch release v0.14.2 today? We would like this at NTNU because #852 fixes a critical bug in a tutorial which will be given at a workshop next week.

Changes since v0.14.1:
* [Changelog](https://github.com/pyxem/pyxem/blob/master/CHANGELOG.rst)
* https://github.com/pyxem/pyxem/compare/v0.14.1...master

pc494, 2022-06-15:
I will start work on this at about 1pm CET. Will see if I can quickly get the last thing I wanted in, and if not, will skip doing that for now and get a patch out.

hakonanes, 2022-06-15:
Thank you! I could also do the release, as I think I have the necessary privileges, but I'll leave it up to you this time.

pc494, 2022-06-15:
Checklist before closing:

- [ ] safe upload to zenodo 
- [ ] " " " pypi
- [ ] " " " conda

pc494, 2022-06-16:
All done :D

closed 2022-06-16 by pc494