pyxem/pyxem #857: Migrate MIB reader to LiberTEM reader

uellue, 2022-06-22:
CC @sk1p for coordination

**Is your feature request related to a problem? Please describe.**

This will allow reading multifile MIB datasets. Furthermore, the current reader in pyxem fails to parse the headers of some of the MIB files we have in our test repertoire for LiberTEM:

<details>

```
pyxem.load_mib(r'C:\Users\weber\Nextcloud\LargeData\ER-C-1\groups\data_science\data\reference\MIB\quad\Au_focused\1-bit_raw-20210728_102044\1-bit_raw.mib')

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Input In [29], in <cell line: 1>()
----> 1 pyxem.load_mib(r'C:\Users\weber\Nextcloud\LargeData\ER-C-1\groups\data_science\data\reference\MIB\quad\Au_focused\1-bit_raw-20210728_102044\1-bit_raw.mib')

File ~\miniconda3\envs\ediff\lib\site-packages\pyxem\utils\io_utils.py:76, in load_mib(mib_path, reshape, flip)
     33 def load_mib(mib_path, reshape=True, flip=True):
     34     """Read a .mib file or an h5 stack file using dask and return as a lazy pyXem / hyperspy signal.
     35 
     36     Parameters
   (...)
     74                     └── signal_type = TEM
     75     """
---> 76     hdr_stuff = _parse_hdr(mib_path)
     77     width = hdr_stuff["width"]
     78     height = hdr_stuff["height"]

File ~\miniconda3\envs\ediff\lib\site-packages\pyxem\utils\io_utils.py:424, in _parse_hdr(fp)
    381 """Parse information from mib file header info from _manageHeader function.
    382 
    383 Parameters
   (...)
    420         number of characters at the header.
    421 """
    422 hdr_info = {}
--> 424 read_hdr = _manageHeader(fp)
    426 # Set the array size of the chip
    428 if read_hdr[3] == "1x1":

File ~\miniconda3\envs\ediff\lib\site-packages\pyxem\utils\io_utils.py:359, in _manageHeader(fname)
    356 shuttertime = float(elements_in_header[10])
    358 if PixelDepthInFile == "R64":
--> 359     bitdepth = int(elements_in_header[18])  # RAW
    360 elif PixelDepthInFile == "U16":
    361     bitdepth = 12

ValueError: invalid literal for int() with base 10: '0.000000E+0'
```

</details>

Migrating to LiberTEM will include support for this MIB flavor, furthermore it allows to pool resources to support more flavors in the future.

**Describe the solution you'd like**

Any LiberTEM dataset can be transformed to a Dask array: https://libertem.github.io/LiberTEM/dask.html#load-datasets-as-dask-arrays That means the "payload" data should be straightforward to port, eliminating the burden to maintain the data decoding functionality twice.

The pyxem MIB reader also attempts some "magic" to guess skipped frames and scan shape if it is not in the metadata. This could be valuable for LiberTEM so that it is available in both projects.

The pyxem reader adds three dead pixels at the boundary of quad layouts. LiberTEM doesn't do that. We should discuss if that is a good default behavior for LiberTEM as well or if it should remain only in pyxem.

LiberTEM should make all header fields available with a stable, public interface that pyxem requires as metadata. Currently the majority are available from an internal interface, but not all. That means users of the LiberTEM reader can benefit from the header and metadata parsing capability.

**Describe alternatives you've considered**

To my knowledge, there are no other Python readers for QD Merlin MIB files besides LiberTEM and pyxem. One could also consider to just leave it as-is or fix the pyxem reader.

uellue, 2022-06-22:
As a larger issue, would it be desirable that ` hs.load(<filename>)` works with MIB or HDR files? Wasn't there a plugin architecture that allows to register readers? Before jumping to fix `load_mib()`, maybe one could discuss first if `pyxem` is even the right place for a HyperSpy compatible MIB reader?

pc494, 2022-06-23:
Hi @uellue, I'm probably not the correct person for this, but from a historical perspective pyxem has a mib reader simply because we needed one at some point, and once it was written we realised we also wanted the flexibility to change it as new problems arose (which is why it's here and not in hyperspy). In an ideal world there would be an IO package for the community, but that has proved hard to realise.

magnunor, 2022-06-23:
Indeed! A `mib` reader was also in the old `pixstem` library, and is in pyxem for pragmatic reasons: a `mib` reader was need to be able to look at the data.

I think a better location for file readers for "generic" file formats such as `mib`, is in an "upstream" package. For example in HyperSpy (or some IO-package). But due to resource constraints, this hasn't really happened. It hasn't been high up on the priority list, since the only thing (at least I) use the the file reader for, is to convert it into a more appropriate format with compression (since transferring and storing uncompressed files is a hassle).

uellue, 2022-06-23:
Regarding dependencies: These are the transitive dependencies that LiberTEM would add compared to a vanilla pyxem installation.

```
async-generator==1.10
autopep8==1.6.0
click==8.1.3
distributed==2022.6.0
empyre==0.3.1
HeapDict==1.0.1
libertem==0.9.2
msgpack==1.0.4
ncempy==1.9.0
Pillow==9.1.1
pycodestyle==2.8.0
sortedcontainers==2.4.0
tblib==1.7.0
toml==0.10.2
zict==2.2.0
```

CSSFrancis, 2022-07-06:
Just a note that with the revival of https://github.com/hyperspy/hyperspy/pull/2972 we should maybe reconsider how to integrate this.  I think that the idea of a separate plugin library for hyperspy is intended to package some of these "extra" file formats. 

There is probably more of a chance of adding specific dependencies for some file loader rather than adding directly to hyperspy. @uellue you might be interested in this as well

closed 2023-11-19 by CSSFrancis