pyxem/pyxem #858: Issue using Pyxem

lnaglecocco, 2022-06-23:
Hi,
I have installed Pyxem but when I try to run it I get issues. Initially I had some numpy issues, but I updated numpy and it was fine. Now I'm getting this:

cannot import name 'register_translation' from 'skimage.feature' (C:\ProgramData\Anaconda3\lib\site-packages\skimage\feature\__init__.py)

I have checked and there is no register_translation within skimage.feature on my computer. I have tried updating both pyxem and skikit-image on my computer and the error still occurs. Does anyone know the solution?

Thanks
Liam

pc494, 2022-06-23:
Hi Liam,

Could you just confirm that you're using scikit-image `0.19.3` and pyxem `0.14.2`? If so it's almost certainly another bug (eg. #786), although I think it's simply a renaming so it should be an easy fix.

lnaglecocco, 2022-06-23:
Hi Phillip,

No, I didn't have these installed. I'd done conda update assuming that
would give me the most recent versions, but it seems it didn't. I've now
manually installed using conda-forge and it is indeed working.

Thanks for the suggestion!

Liam

pc494, 2022-06-23:
Sorry, I'm just going to reopen this as I'm pretty sure it means we need tighter scikit-image bound, say 0.19 and upwards. @CSSFrancis you worked on #786 does that seem correct to you?

CSSFrancis, 2022-06-24:
> Sorry, I'm just going to reopen this as I'm pretty sure it means we need tighter scikit-image bound, say 0.19 and upwards. @CSSFrancis you worked on #786 does that seem correct to you?

I think so, I'll have to look into it a little bit more next week. I'm on vacation this week with little to no service.

pc494, 2022-06-26:
@CSSFrancis no rush, enjoy your holiday; just wanted to make sure we join the two issues up.

CSSFrancis, 2022-07-14:
@lnaglecocco Sorry for the delay.  I got married recently and was on my honeymoon. 

What version of pyxem are you using? I would suggest updating that first and I think that should work for you.  If you need to use a  an older version of pyxem for some reason you might need to downgrade skimage to something before March 2020 https://github.com/scikit-image/scikit-image/pull/4502

pc494, 2022-07-14:
Congratulations @CSSFrancis :champagne: 

In terms of this issue I think I am happy to close it. Perhaps our min version for skimage in v0.14 isn't right but we can deal with that separately.

closed 2022-07-14 by pc494