pyxem/pyxem #866: Different behaviour of "find_beam_offset_cross_correlation" after scikit-image 0.19.0

PVacek, 2022-07-26:
Dear pyxem community,
I found that behaviour of `find_beam_offset_cross_correlation` changed after scikit-image update (from version 0.19.0). See quote from release notes below. From my experience the new version does not work as well as the old one and I rather use scikit-image version 0.18.3 to get the old behaviour. 
As it is described in release notes, it can be "fixed" by specifying `normalization=None` in `phase_cross_correlation` parameters. Sadly, `find_beam_offset_cross_correlation` does not support passing this argument to `phase_cross_correlation` function. Can this option be added in upcomming versions of pyxem?

From scikit-image release notes. (https://scikit-image.org/docs/stable/release_notes.html)

> A bug fix to the phase normalization applied within skimage.register.phase_cross_correlation may result in a different result as compared to prior releases. The prior behavior of “unnormalized” cross correlation is still available by explicitly setting normalization=None. There is no change to the masked cross-correlation case, which uses a different algorithm.

Petr

CSSFrancis, 2022-07-26:
That is a good catch! Thanks for the bug report.  

I was planning on redoing how pyxem handles cross correlations thoughout so that it is more consistant.  The phase_cross_correlation function in skimage is a little strange especially when considering masks so I currently have a PR https://github.com/scikit-image/scikit-image/pull/6077 to fix a lot of that functionality in skimage. That should trickle down to pyxem and stabilize a lot of that code.

This is a good reminder to me to get that all sorted and committed so that we can get this sorted in the next version of pyxem. It will also allow us to use masks for all of these functions and to better handle if we want to normalize that data (and how we want to normalize the data.

pc494, 2022-08-18:
I think we should copy+paste the 0.18 implementation into pyxem as a short term solution. @CSSFrancis do you think that makes sense?