pyxem/pyxem #867: Implement Logging in pyxem

CSSFrancis, 2022-08-10:
**Is your feature request related to a problem? Please describe.**
I think that we should follow hyperspy in using logging to throw errors, information or Warnings. 

**Describe the solution you'd like**
I will start to implement this when adding deprecation warnings. 

**Describe alternatives you've considered**
@hakonanes Do you have any thing else to add?  I was going to copy the deprecation warning from kikuchipy as well to implement into pyxem.

pc494, 2022-08-11:
I agree broadly with the top answer on the linked SO question. `Warnings` for client should consider doing something (including deprecations), logging for developers (we) should do something.

hakonanes, 2022-08-11:
Yes, please go ahead and copy the deprecation utilities and tests, keeping the links to Matplotlib and scikit-image, where we originally took it from in kikuchipy/orix (orix' utilities were modified by @harripj).

CSSFrancis, 2022-08-17:
This should be partially covered by #868.  I would like to implement logging similar to hyperspy as well before the 1.0.0 release.

hakonanes, 2022-11-11:
I added logging to *kikuchipy* a couple months ago:
* Implementation and tests: https://github.com/pyxem/kikuchipy/commit/86d2b543e6ed9645a7301e44c30153619c744085
* Only use so far: https://github.com/pyxem/kikuchipy/blob/41b2409b9a4c18d7c5573cf0160bc380159412bc/kikuchipy/signals/_kikuchipy_signal.py#L358-L407

I added logging to ease debugging of some overwriting of *HyperSpy* `Signal2D` methods. To start using logging in *pyxem*, I suggest to start small like this when there is a particular need for it, and then use it more where necessary.

CSSFrancis, 2022-11-12:
@hakonanes Once again it seems like we might be copying you :) 

I know I opened this thread but my opinion on logging is complicated. I like the idea of logging but often get frustrated by it. Specifically, I don't like having warnings visible (and common) because they can scare new users. Many times when training new people they see something red and freeze. I'm afraid that lots of new users see this and panic. I'd love a way to fix this. Maybe adding a note at the end of each warning saying to suppress this warning use `hs.set_logging_level("ERROR")`.

My other thought is what is your reason for not piggy backing off of hyperspy?  It might be nice to set it once for every package. 
Or have hyperspy set for all sub packages. I like the idea of setting the logging level once rather than individually.

hakonanes, 2022-11-12:
> we might be copying you :)

Which means pyxem will copy HyperSpy, since kikuchipy copied them!

> I don't like having warnings visible (and common) because they can scare new users.

Valid point. I use warnings to in situations like when a user obtains a limited results due to a file reader not reading everything it expected to find in a file, or when some functionality is deprecated and they should switch to something else. I don't see another way of informing users of these things than warnings.

> My other thought is what is your reason for not piggy backing off of hyperspy

When I was debugging, I wanted to kikuchipy's debug logs, not HyperSpy's. I'm not sure how common it is for one package to depend on another package's logging. Haven't looked into it., but is it even possible? I still prefer to have the option to set both package's log levels independently.