pyxem/pyxem #870: Current build failure due to .apply_affine_transform()

pc494, 2022-08-15:
The failure (from the CI) occurs at:

```python
   def test_apply_affine_transforms_paths(self, diffraction_pattern):
        D = np.array([[1.0, 0.9, 0.0], [1.1, 1.0, 0.0], [0.0, 0.0, 1.0]])
        s = Signal2D(np.asarray([[D, D], [D, D]]))
        static = diffraction_pattern.apply_affine_transformation(D, inplace=False)
        dynamic = diffraction_pattern.apply_affine_transformation(s, inplace=False)
```

and is probably due to a change in skimage versioning.

pc494, 2022-08-15:
When I get to this I'll try and pick up #840, #841 and #856 as well.

pc494, 2022-08-16:
and I think I'll make a release so that #868 can come into main as soon as possible

CSSFrancis, 2022-08-16:
I looked into this as well, it seems the error only occurs when the transform is iterated alongside the signal in the map function.  Not sure exactly what causes this but this might be an upstream hyperspy problem as well.

CSSFrancis, 2022-11-08:
Realized I never fixed this one.  The problem is related to the `np.squeeze` function in `map_iterate`.  This calls np.asarray which tries to convert a function to an array.  From there it gets very confused. For now we can by pass this just by manually setting the output dtype and signal size.  This is also a little bit of a speedup too.

closed 2022-11-08 by CSSFrancis