pyxem/pyxem #876: Use transpose() instead of deprecated set_signal_dimension()

hakonanes, 2022-09-25:
These changes are necessary as `set_signal_dimension` is deprecated: https://hyperspy.org/hyperspy-doc/current/user_guide/changes.html#deprecations

See warning in the test suite:
```python
pyxem/tests/generators/test_displacement_gradient_tensor_generator.py: 8 warnings
pyxem/tests/generators/test_integration_generator.py: 3 warnings
pyxem/tests/generators/test_pdf_generator.py: 3 warnings
pyxem/tests/generators/test_subpixelrefinement_generator.py: 1 warning
pyxem/tests/generators/test_virtual_image_generator.py: 4 warnings
pyxem/tests/signals/test_diffraction_vectors.py: 35 warnings
pyxem/tests/signals/test_segments.py: 18 warnings
pyxem/tests/signals/test_strain_map.py: 10 warnings
pyxem/tests/signals/test_tensor_field.py: 1 warning
pyxem/tests/signals/test_virtual_dark_field_image.py: 4 warnings
pyxem/tests/utils/test_io_utils.py: 2 warnings
  /opt/hostedtoolcache/Python/3.8.13/x64/lib/python3.8/site-packages/hyperspy/axes.py:2[124](https://github.com/pyxem/pyxem/actions/runs/2895463648/jobs/4605689078#step:8:125): VisibleDeprecationWarning: Using `set_signal_dimension` is deprecated, use `as_signal1D`, `as_signal2D` or `transpose` of the signal instance instead.
    warnings.warn(("Using `set_signal_dimension` is deprecated, use "
```

_Originally posted by @ericpre in https://github.com/pyxem/pyxem/issues/874#issuecomment-1257035518_
      
Thank you, @boneyag, for opening #874 to shed light on this necessary change.

I'm not familiar with the `DiffractionVectors` class, but changing from calls `s.axes_manager.set_signal_dimension()` inplace to `s2 = s.transpose(signal_axes=0)` does not work since the latter does not return a `DiffractionVectors`. Either `DiffractionVectors` wraps HyperSpy's `transpose()`, or the signal returned from `transpose()` must be set to `DiffractionVectors` correctly.

pc494, 2023-01-19:
or by #898? @CSSFrancis can you comment

CSSFrancis, 2023-01-19:
> or by #898? @CSSFrancis can you comment

Yep this is closed now

closed 2023-01-19 by CSSFrancis