pyxem/pyxem #877: Test suite failure lazy signal

ericpre, 2022-10-01:
From https://github.com/hyperspy/hyperspy-extensions-list/actions/runs/3162982508/jobs/5150109391 - most likely related to a `dask` update.

```python
_______________________ TestGetChunking.test_chunk_bytes _______________________

self = <pyxem.tests.utils.test_dask_tools.TestGetChunking object at 0x7f43a0f583a0>

    def test_chunk_bytes(self):
        s = LazyDiffraction2D(da.zeros((32, 32, 256, 256), dtype=np.uint16))
        chunks = dt._get_chunking(s, chunk_bytes="15MiB")
>       assert chunks == ((8, 8, 8, 8), (8, 8, 8, 8), (256,), (256,))
E       assert ((10, 10, 10,...256,), (256,)) == ((8, 8, 8, 8)...256,), (256,))
E         At index 0 diff: (10, 10, 10, 2) != (8, 8, 8, 8)
E         Full diff:
E         - ((8, 8, 8, 8), (8, 8, 8, 8), (256,), (256,))
E         ?   ^  ^  ^  ^    ^  ^  ^  ^
E         + ((10, 10, 10, 2), (10, 10, 10, 2), (256,), (256,))
E         ?   ^^  ^^  ^^  ^    ^^  ^^  ^^  ^

/usr/share/miniconda3/lib/python3.10/site-packages/pyxem/tests/utils/test_dask_tools.py:584: AssertionError
```

hakonanes, 2022-10-01:
As reported by @ericpre in https://github.com/pyxem/kikuchipy/issues/565, this is an issue in kikuchipy's Dask chunking functions as well (since I adapted them from the functions in pyxem). I hope to look into it in the next few days, but would be happy to hear of the fix if anyone else does so before me.

hakonanes, 2022-10-03:
Made a fix in #878.

closed 2022-10-03 by pc494