pyxem/pyxem #879: Method " dp.get_integrated_intensity(roi)" Not working properly after hyperspy update from 1.6.5 to 1.7.x

garrofederico, 2022-10-14:
**Describe the bug**
Dear pyxem community,

After the update  hyperspy update from 1.6.5 to after 1.7.0 the dp.get_integrated_intensity(roi) is not working properly. 

functions like dp.get_integrated_intensity(roi).plot() or hs.plot.plot_images(vdf_stack) are plotting an empty image (only white value pixels). After further investigating the bug, I found that the dp.get_integrated_intensity(roi) is returning a signal with a .data attribute made of a numpy array filled with NaN values, so there seems to be an issue at the level of the _get_sum_signal() function in the [common_diffraction.py]  (https://github.com/pyxem/pyxem/blob/39693eba7c1d7d2859fe43278e436b5a93a7ad6c/pyxem/signals/common_diffraction.py) script. Maybe is the way in which signal.nansum() is handled after the hyperspy update?

```
def _get_sum_signal(signal, out_signal_axes=None):
        **_out = signal.nansum(signal.axes_manager.signal_axes)_**
        if out_signal_axes is None:
            out_signal_axes = list(
                np.arange(min(signal.axes_manager.navigation_dimension, 2))
            )
        if len(out_signal_axes) > signal.axes_manager.navigation_dimension:
            raise ValueError(
                "The length of 'out_signal_axes' can't be longer"
                "than the navigation dimension of the signal."
            )
        # Reset signal to default Signal1D or Signal2D
        out.set_signal_type("")
        return out.transpose(out_signal_axes)
```
in hyperspy version  1.7.0 The function is working correctly with .mib files in format, but not with .zspy. from 1.7.1 and beyond not even .mib files are working properly.

The method dp.plot_integrated_intensity(roi) is plotting and image (even though I'm not sure that the values of this image are correct, as with dp.get_integrated_intensity(roi) method) ,but I cannot save it in a variable for later use.

**To Reproduce**
Steps to reproduce the behavior:
1. Run dp.get_integrated_intensity(roi) method with hyperspy version 1.6.5:
```
dp = pxm.load_mib("path/to/data...")
...
dp.get_integrated_intensity(roi).plot()
```
2. Run same code with hyperspy version 1.7.0/ 1.7.1/1.7.2
3. See error

**Expected behavior**
The code should work in the same way, regardless of hyperspy version.

**Why this is important**
The 1.7.0 version of hyperspy is important for me because I need to work with .zspy files, which is one of the features of this release: http://hyperspy.org/hyperspy-doc/current/user_guide/changes.html

**Additional context**

For context I'm using pyxem version 0.14.1
`pyxem                     0.14.1             pyhd8ed1ab_0    conda-forge`

and datasets that range the 100Gb in size. Maybe it is related to this [change](https://github.com/pyxem/pyxem/blob/master/CHANGELOG.rst)?
> Changed
> For developers: HyperSpy's .map function will now be used to process big datasets, instead of pyXem's process_dask_array

CSSFrancis, 2022-10-14:
This is an interesting bug...  The way that the `get_integrated_intensity` function works is that it just calls the roi on the data and then sums the data.

You can repeat this buy running:

```
mask = hs.roi.CircleROI(10.0,10.0, 20.0) # note that these should be in real units and not pixel units
virtual_img = mask(s) # slices the signal with the ROI.  This masks are area outside of the circle with `np.nan`
virtural_img.nansum(virtural_img.axes_manager.signal_axes) # sum up signal axes

# should give the same result as...

s.get_integrated_intensity(mask)
```

I tested both ways of doing this with a 50 GB dataset and it seems to be operating correctly.

One check would be to make sure that your roi is actually behaving as you think it is. I would maybe plot the first frame of the virtual image to make sure that you roi doesn't have a radius of essentially 0 as this could cause the bug you are seeing as well. 

Maybe try upgrading pyxem to 0.14.2?  The relevant changes are #722 which should be implemented in 0.14.1 but that might be safer.

CSSFrancis, 2022-10-14:
As to the function working differently with different file types that suggests that maybe your axes managers are not equal for the `MIB` and `.zspy` files?  

You could also try using the default and distributed dask schedulers (I tried using both and they should both work with the `zsp` format.  This won't work with the `.MIB` file type due to how we are currently loading that data. I would recommend the [distributed](https://distributed.dask.org/en/latest/) scheduler if you can get it to work on your set up.  Mostly because you can use the [dashboard](https://docs.dask.org/en/stable/dashboard.html) which is very helpful for debugging/ telling if you code is running properly or not. It is also more aware of your memory/CPU limitations as well as splitting and sharing tasks. 

Let me know if you have any other problems, especially related to using `.zspy` and processing large datasets. I've tested up to a couple of Tb sized datasets so it should scale to your needs but there are some pitfalls which can be frustrating when scaling up.

CSSFrancis, 2022-11-02:
@garrofederico Did you ever figure out what the issue was here?