pyxem/pyxem #880: Failure with numpy dev

ericpre, 2022-10-31:
From https://github.com/LumiSpy/lumispy/issues/157

```python
=================================== FAILURES ===================================
_____________ test_profile_indexation_generator_single_indexation ______________

profile_simulation = <diffsims.sims.diffraction_simulation.ProfileSimulation object at 0x7f8f87e74640>

    def test_profile_indexation_generator_single_indexation(profile_simulation):
        pig = ProfileIndexationGenerator(
            magnitudes=[
                0.31891931643691351,
                0.52079306292509475,
                0.6[106](https://github.com/hyperspy/hyperspy-extensions-list/actions/runs/3362966097/jobs/5575418400#step:17:107)839974876449,
                0.73651261277849378,
                0.80259601243613932,
                0.9020400452156796,
                0.9567579493[107](https://github.com/hyperspy/hyperspy-extensions-list/actions/runs/3362966097/jobs/5575418400#step:17:108)4043,
                1.0415861258501895,
                1.0893168446141808,
                1.1645286909[108](https://github.com/hyperspy/hyperspy-extensions-list/actions/runs/3362966097/jobs/5575418400#step:17:109)374,
                1.2074090451670043,
                1.2756772657476541,
            ],
            simulation=profile_simulation,
        )
>       indexation = pig.index_peaks(tolerance=0.02)

/usr/share/miniconda3/lib/python3.10/site-packages/pyxem/tests/generators/test_indexation_generator.py:190: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
/usr/share/miniconda3/lib/python3.10/site-packages/pyxem/generators/indexation_generator.py:232: in index_peaks
    return index_magnitudes(np.array(self.magnitudes), self.simulation, tolerance)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

z = array([0.31891932, 0.52079306, 0.610684  , 0.7365[126](https://github.com/hyperspy/hyperspy-extensions-list/actions/runs/3362966097/jobs/5575418400#step:17:127)1, 0.80259601,
       0.90204005, 0.95675795, 1.04158613, 1.08931684, 1.16452869,
       1.20740905, 1.27567727])
simulation = <diffsims.sims.diffraction_simulation.ProfileSimulation object at 0x7f8f87e74640>
tolerance = 0.02

    def index_magnitudes(z, simulation, tolerance):
        """Assigns hkl indices to peaks in the diffraction profile.
    
        Parameters
        ----------
        simulation : DiffractionProfileSimulation
            Simulation of the diffraction profile.
        tolerance : float
            The n orientations with the highest correlation values are returned.
    
        Returns
        -------
        indexation : np.array()
            indexation results.
    
        """
        mags = z
        sim_mags = np.array(simulation.magnitudes)
        sim_hkls = np.array(simulation.hkls)
        indexation = np.zeros(len(mags), dtype=object)
    
        for i in np.arange(len(mags)):
            diff = np.absolute((sim_mags - mags.data[i]) / mags.data[i] * 100)
    
            hkls = sim_hkls[np.where(diff < tolerance)]
            diffs = diff[np.where(diff < tolerance)]
    
            indices = np.array((hkls, diffs))
>           indexation[i] = np.array((mags.data[i], indices))
E           ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (2,) + inhomogeneous part.

/usr/share/miniconda3/lib/python3.10/site-packages/pyxem/utils/indexation_utils.py:[140](https://github.com/hyperspy/hyperspy-extensions-list/actions/runs/3362966097/jobs/5575418400#step:17:141): ValueError
_________________________ test_calculate_norms_ragged __________________________

    def test_calculate_norms_ragged():
>       norms = calculate_norms_ragged(np.array([[3], [6, 8]]))
E       ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (2,) + inhomogeneous part.

/usr/share/miniconda3/lib/python3.10/site-packages/pyxem/tests/utils/test_vector_utils.py:38: ValueError
```

CSSFrancis, 2022-11-01:
@ericpre I'll fix this tomorrow. Just need to add the `type=object` to the array creation.

closed 2022-11-04 by CSSFrancis

ericpre, 2022-11-05:
Thanks @CSSFrancis, it fixes https://github.com/hyperspy/hyperspy-extensions-list/pull/28 - run: https://github.com/hyperspy/hyperspy-extensions-list/actions/runs/3398315781.