pyxem/pyxem #887: Update the default branch to 'main'

pc494, 2023-01-10:
Almost all documentation now uses 'main' over 'master' after the policy change in 2022. It's really easy to do the update (online) and the local change is also not very difficult, so unless I hear objections I'm going to push on with that.

CSSFrancis, 2023-01-10:
That sounds like a good thing going forward.  Go for it!

pc494, 2023-01-11:
done, GitHub gave me the following information:

"""
Your members will have to manually update their local environments. We'll let them know when they visit the repository, or you can share the following commands.

```
git branch -m master <BRANCH>
git fetch origin
git branch -u origin/<BRANCH> <BRANCH>
git remote set-head origin -a
```
"""

closed 2023-01-11 by pc494

hakonanes, 2023-01-13:
We should update the references to the `master` branch in the contributing guide as well: https://pyxem.readthedocs.io/en/latest/contributing.html#making-changes