pyxem/pyxem #891: Version 0.15 Release Tracker

CSSFrancis, 2023-01-11:
**Version 0.15 Release Tracker**
--------------------------------------------

It has been a bit over 6 months from our last bug patch and longer since our last minor release.  This will be the last minor release before the major 1.0.0 release.   Most of the changes in this release are visible deprecation warnings before the 1.0.0 release.

I plan on going though every file in the package with the express purpose of rewriting code which isn't  fully compatible with fully lazy operations.  Mostly this means moving every function to use the `hyperspy.BaseSignal.map` function if possible.   I also want to formalize the different signals and reduce the amount of duplicated code, especially code that is duplicated here and upstream in hyperspy. 

As much as possible I would like to reduce the extent of the tests as well.  This means removing tests which are dependent on hyperspy and test things like passing axes which is already tested upsteam in hyperspy. 

**Release Tracker**
---------------------------

- [ ] #866 
- [ ] #889 
- [ ] #886 
- [x] #868

pc494, 2023-01-11:
Great stuff @CSSFrancis, I would be keen to help with bits of this, as well as maybe contributing to some code purging to help keep the repository as lean as possible,

CSSFrancis, 2023-01-17:
I've realized that I made some changes associated with the 1.0.0 release that should probably be applied to the 0.15 release.  I will make a new PR cherry picking the commits that need to be back tracked.

hakonanes, 2023-01-18:
By the way, it is very important for all these new changes leading up to a 0.15 release (removal, changes to API, bug fixes, additions) to be documented in the changelog! Otherwise users will have greater difficulties figuring out why their existing code fails with v0.15.

hakonanes, 2023-01-20:
@CSSFrancis and @pc494, do you anticipate a sufficiently common workflow to be broken with version 0.15?

If so, might it be an idea to make a release candidate, named 0.15.0rc1 (according to [convention](https://peps.python.org/pep-0440/#pre-releases)), for some people to try before releasing 0.15.0? After such an RC is released, people doing `pip install pyxem` will get the latest stable release (0.14.2) while the RC can be installed with `pip install --pre pyxem` (or `pip install pyxem==0.15.0rc1`). This way possible bugs can be fixed before releasing 0.15.0, potentially avoiding several patch releases.

pc494, 2023-01-23:
I'm expecting a release candidate as I think it would be good to have run all the demos before we ship.

hakonanes, 2023-01-23:
> it would be good to have run all the demos before we ship

Definitely. I will try to notify users at NTNU as well when a release candidate is available, so that people can report any bugs or inconsistencies.

CSSFrancis, 2023-03-08:
My plan is to try to push this out tomorrow.  It would be nice to include #889 but I think that fixing #905 is more important and we can always push a patch later as there isn't much of a change other than some underlying stuff that specifically makes working with big datsets break. 

The fixes to strain mapping are also important and should be released sooner rather than later.

CSSFrancis, 2023-03-16:
I think we should be good to make a 0.15 release tomorrow unless people have a reason against it

pc494, 2023-03-19:
Sorry to have been absent in these final stages, changing jobs put more work on my plate than I expected. Thanks both for your efforts, look forward to seeing 0.15 out in the wild very soon!

CSSFrancis, 2023-03-19:
@pc494 no problem! Everything seems to happen at once some times :)  As far as making the release I am going to be honest I remember we talked about it a little while back but we have to manually add pyxem to pypi or should be github action work to do it automatically. 

There are a couple small changes in #917 but I think beyond that this should be ready to ship.

hakonanes, 2023-03-21:
Would be nice if pyxem had a release guide, similar to kikuchipy's (https://github.com/pyxem/kikuchipy/blob/develop/RELEASE.rst).

CSSFrancis, 2023-03-21:
@hakonanes lets hold this one more day! I need to fix some bugs with the example notebooks when I ran through them last night. I'll make a PR with the changes and then we can go from there.

CSSFrancis, 2023-04-06:
We should be released now.  I just need to make sure that the conda-forge bot comes and makes that release.

As far as zenodo goes is there something that we need to do there?

pc494, 2023-04-06:
> As far as zenodo goes is there something that we need to do there?

should be automatic, will check the integration worked tomorrow, but right now I would guess it's just a little slow.

CSSFrancis, 2023-04-06:
It seems like it ran but I think you'll have to review/ merge it. :)

https://github.com/conda-forge/pyxem-feedstock/pull/24

hakonanes, 2023-04-06:
I'll see to the conda-forge release tomorrow Friday. We should also fill out the [release notes](https://github.com/pyxem/pyxem/releases/tag/v0.15.0). Feel free to do that, @CSSFrancis.

Thanks for making this release happen!

hakonanes, 2023-04-06:
@CSSFrancis, also, nice touch on the binder badge in the demos repo. And very good that you had a look at them for this release. But perhaps Binder should point to the pyxem demos repo and not your fork?

CSSFrancis, 2023-04-06:
> @CSSFrancis, also, nice touch on the binder badge in the demos repo. And very good that you had a look at them for this release. But perhaps Binder should point to the pyxem demos repo and not your fork?

Huh that wasn't intentional. Somewhere along the way the readme got lost with the notebooks and I jut copied what used to be there.  Apparently I had been using my fork to serve the binder badge.  

I'll look at that again.  Pretty sure the environment information got removed as well so that should be updated too.

hakonanes, 2023-04-07:
> I'll see to the conda-forge release tomorrow Friday.

I cannot merge the pyxem feedstock as I'm not a maintainer of it, I realized. Sorry about that. Either @pc494, @dnjohnstone or @magnunor has to do it.

hakonanes, 2023-04-07:
> Pretty sure the environment information got removed as well so that should be updated too.

Yes, the Binder link does not currently work. Should add an `environment.yml` file recognized by MyBinder to the demos repo.

CSSFrancis, 2023-04-07:
> > Pretty sure the environment information got removed as well so that should be updated too.
> 
> Yes, the Binder link does not currently work. Should add an `environment.yml` file recognized by MyBinder to the demos repo.

Yea I started that yesterday. https://github.com/CSSFrancis/pyxem-demos/tree/AddEnviornment The environment currently doesn't run though because of a numba-numpy version disagreement.  Kind of weird because I would assume that conda would handle that.  https://github.com/numba/numba/issues/8615

CSSFrancis, 2023-04-07:
@hakonanes  The binder links should be fixed now.  We just need to get pyxem pushed to conda-forge for them to be fully operational.

hakonanes, 2023-04-07:
Great! (Note that one can ask for pip installations in the environment file ([example](https://github.com/pyxem/kikuchipy/blob/develop/environment.yml)).)

pc494, 2023-04-10:
I think the Zenodo might have failed. I suggest we open that as a separate issue and close this.

closed 2023-04-13 by hakonanes