pyxem/pyxem #892: Small Changes to the Strain Mapping Routine

CSSFrancis, 2023-01-12:
I recently tried to do some strain mapping and found the code currently in pyxem to be lacking in some places.  In fact it appears to be broken. 

I've made some small changes to the code and plan on making some bigger ones down the line for the RELEASE_next_major

**Is your feature request related to a problem? Please describe.**

1 .The code for transforming peaks to diffractionVectors doesn't work. This is a pretty glaring problem.
2. There is no way to filter peaks to only the ones you are interested in.  This will get much better in the future but it is 
still something that we should be working on. 
3. The Example notebook for strain mapping is pretty woefully incomplete.
4. The transpose function when calculating the Displacement gradient function is not necessary given the change to how hyperspy returns peak positions.
5. The tests for strain mapping don't test some of the functionality.  They don't properly account for these changes. 

I'll start working through these changes.  I've got a working jupyter notebook and a set of code changes that should make this a little more clear. 

@pc494  if you have any input I would be happy to hear it.   I've done fairly limited strain mapping experiments but I think I have a fairly good handle on how the experiments are set up.

pc494, 2023-01-13:
Yeah this is me. I think it would probably be best if we have a meeting and report the content here. Any of the emails you have for me should work :)

CSSFrancis, 2023-01-13:
> Yeah this is me. I think it would probably be best if we have a meeting and report the content here. Any of the emails you have for me should work :)

Sounds good.  I just sent you an email.  I'm pretty much going to be working on this for the next week full time until I need to work on some other projects.

PR #895 (2023-01-17): Bugfix for Strain Mapping Code
---
name: Bugfix for Strain Mapping Code
about: This code fixes the broken strain mapping code and simplifies the workflow

---

For a more complete example of how this works look at:
https://github.com/pyxem/pyxem-demos/pull/80

**Checklist**
- [ ] Updated CHANGELOG.md
- [ ] (if finished) Requested a review (from pc494 if you are unsure who is most suitable)

**What does this PR do? Please describe and/or link to an open issue.**
Simplifies the work flow:

1. Filter data/ subtract background etc.
2. Find peaks in the data
3. Refine positions (need to add subpixe [... cut]
closed 2023-01-17 by pc494