pyxem/pyxem #894: [deletion candidate] investigate_dog_background_removal_interactive

pc494, 2023-01-13:
https://github.com/pyxem/pyxem/blob/0779774368e8f468c9317fe616fb331ba1f61984/pyxem/utils/expt_utils.py#L679

This is an okay function, but it's too niche to be part of the package. Candidate for removal in v0.15.0

hakonanes, 2023-01-13:
Do you mean to deprecate or remove it in v0.15? There might be someone using it out there... One solution is to, in a deprecation warning, ask users of it to say so on GitHub. We've done this in orix ([example](https://github.com/pyxem/orix/pull/392/files#diff-896090017cd74ebcb6f0b90eb7e1328e74e9a425ac5af40d22b7d829df5e8ae3R762-R769)).

pc494, 2023-01-13:
Sorry, this was badly worded, deprecation warning in 0.15.0, and removal in 1.0.0

pc494, 2024-01-26:
Will pick up in #994

pc494, 2024-01-29:
Closed by #994

closed 2024-01-29 by pc494