pyxem/pyxem #899: Refactor Vector Orientation Mapping Code

CSSFrancis, 2023-01-17:
I think that the Vector Orientation Mapping Code has some fairly large bugs which have become somewhat of a problem.  I think that we should remove much of that code going into the 1.0.0 release.  I do think that Vector Orientation Mapping, however, is something that we should support.

With the addition by @din14970 related to accelerated pattern matching I think that there is the ability to take some of that code and use that to write a simplified version using only the sparse vectors found. 

The workflow would go something like:

1. Find the diffraction spots--> Transform the diffraction spots to polar coordinates. 
2. Simulate the patterns for the entirety of the dataset and compute the polar unwrapped version. (Similar to what is currently done)
3. Calculate the normalized cross correlation by just multiplying those sparse points across the simulated patterns. 
4. Score and refine using the same methods as the other method.

The only real change would be writing a new cross-correlation method which should be fairly minor.  We could probably do this all without any lazy operation and in most cases this might all fit on the GPU which would be fairly fast.

CSSFrancis, 2023-01-17:
I'll write up a cross-correlation method tomorrow

hakonanes, 2023-01-18:
This sounds very good!

I know the template matching code (mostly Niels' [@din14970]), but there are no automatic benchmarks or measure of memory use as far as I know. This means that we have to be careful if any of his optimized template matching code is refactored. It might actually be safest to duplicate some lines instead of adding conditionals and the like in functions.

I hope we can fit all indexing results in an orix [CrystalMap](https://orix.readthedocs.io/en/stable/reference/generated/orix.crystal_map.CrystalMap.html#orix.crystal_map.CrystalMap)! This map is convenient for quick validation of the results (orientation maps) and saving the results to formats importable in MTEX or commercial softwares (or back into Python) for further processing (grain reconstruction, texture analysis etc.). We already have a function to convert the template matching results into a crystal map ([code](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/indexation_utils.py#L1640)). This function is safe to adjust to fit the results from vector matching. I can help review this part if/when we get to it.

din14970, 2023-01-18:
Do you mean with 2) to simulate patterns sampling all of SO(3) or will you apply the same logic as in the current template matching approach, i.e. optimizing the in-plane angle during the template matching step?

In the former case, I think the simulation in diffsims needs to be substantially sped up. I've been sort of working on this on and off but have found it difficult to find time.

In both cases, I'd be interested to see how you deal with differing numbers of diffraction spots in image vs template respectively, and how you deal with the spatial information of those points. If you just represent both template and image data in a sparse matrix of the same size, you will get extremely noisy results because non-zero pixels will rarely if ever coincide and so you will likely get 0 for most cross correlation calculations.

I think a more promising approach to deal with point-based comparisons is to make use of cKDtrees somehow. This allows you to query nearest template points for each real spot in the image, and then you can calculate some error metric based on average distances and intensities. In the case you make simulations for all of SO(3) I could see a way to implement this. In case you make simulations based on sampling S2, I think you would have to do the same but for each template iterate through the possible in-plane angles.

Note that the slowest part of this approach will probably be finding the diffraction spots in each image.

din14970, 2023-01-18:
Also note that a cKDtree, point-distance based error metric approach will not be "fair" in polar coordinate space because the relative scale between r and theta is arbitrary. For a distance based approach, probably keeping things in cartesian coordinates is best, and if finding in-plane rotation is necessary to just multiply coordinates by a pre-computed rotation matrix.

CSSFrancis, 2023-01-18:
@hakonanes  and @din14970 thanks for the insight!

> I know the template matching code (mostly Niels' [@din14970]), but there are no automatic benchmarks or measure of memory use as far as I know. This means that we have to be careful if any of his [...]

For the most part I think bypassing many of the optimization is right thing to do.  If we can transfer the whole thing to the GPU in one pass this should be extremely fast.  Even on a cpu this should be very fast so it might not even matter too much.

> Do you mean with 2) to simulate patterns sampling all of SO(3) or will you apply the same logic as in the current template matching approach, i.e. optimizing the in-plane angle during the template matching step?

I was referring to the current template matching approach, just comparing vector positions.

> In the former case, I think the simulation in diffsims needs to be substantially sped up. I've been sort of working on this on and off but have found it difficult to find time.

I think that there is some potential for simulating using real intensities, but would argue that any further development should probably move towards using something like a Prismatic simulation/ABTEM.  At the most our Kinematic simulation might need some work dealing with thicker Ewald spheres and some speed improvments. 

> In both cases, I'd be interested to see how you deal with differing numbers of diffraction spots in image vs template respectively, and how you deal with the spatial information of those points. If [...]

Yes the noise is a potential problem.  I do like the idea of converting the points to polar coordinates and then have a cross-correlation function somewhat like this:

```python

def polar_vector_cross_correlate(polar_vectors, template_vectors, cross_correlation_profile, max_r):
    delta_r = polar_vectors[:,0]-template_vectors[:,0] # first column (r)
    delta_t = polar_vectors[:,1]-template_vectors[:,1] # second column (theta)
    # third column is intensity.
    
    # reflect delta theta
    delta_t = np.abs(delta_t)
    delta_t[delta_t>180]= delta_t-180

    # compute cross_correlation
    in_range = delta_r<max_r
    all_profiles = cross_correlation_profile[delta_r[in_range]] *polar_vectors[:,2][in_range]*template_vectors[:,2][in_range]
    all_profiles = strided_index_roll(all_profiles, delta_t[in_range]) # roll axes 
    return np.sum(all_profiles, axis=0)
```

The idea being that you take two flat beams and take the cross correlation in r and theta over some small region to get a cross_correlation profile.  Then you calculate a r and theta displacement for each.  From here you can first identify if the beams will interact and secondly identify the correlation profile/intensity.   Those profiles are then shifted by the offset in theta giving you the cross_correlation.  This is equivalent to taking the two points and making flat disks, convolving them and reading the convolution from the strip in theta at some point r. 

This method is quite flexible as well.  For example you can adjust the interplanar spacing to see if your calibration is slightly off. That just requires another correlation dimension to optimize. As long as the difference is small that should be doable. 

> I think a more promising approach to deal with point-based comparisons is to make use of cKDtrees somehow. This allows you to query nearest template points for each real spot in the image, and then [...]

This would work as well but I like the idea of polar coordinates for speed/ simplicity purposes. The nice thing about polar coordinates is that the r coordinate is really the only coordinate that matters. The distance in theta is only an offset. 

> Note that the slowest part of this approach will probably be finding the diffraction spots in each image.

For the most part I think that this is okay.  For 1 this allows people a way to work with larger datasets and not be too overwhelmed.  It also opens up this kind of analysis to people without extensive GPU resources.  I'm not sure that it will be better (mostly it really depends on how well we identify diffraction spots) but it will offer a different workflow. Finding peaks is already fairly fast.  I was also going to see about accelerating that using GPU's as well.

CSSFrancis, 2023-01-19:
> > the current template matching approach, just comparing vector positions.
> 
> Not sure which one that would be ...

Your implementation takes the position of the calculated diffraction spots in polar coordinates and then multiplies the r position of the diffraction spots by the strip in theta at that r to get the cross correlation. Do I have that correct?  This is just taking that method one step father where both the match and the calculated diffraction spot are vectors. 

> > I think that there is some potential for simulating using real intensities, but would argue that any further development should probably move towards using something like a Prismatic simulation/ABTEM.
> 
> Before I delved into this template matching business I also thought this, but as it turns out, actually "accurate" intensity calculations for template matching don't help much at all. Multiple reasons for this which I describe in the discussion part of the pyxem paper. Edgar Rauch also discovered and described this already almost 20 years ago. One simple reason is that the cross correlation is mostly determined by the most intense diffraction spots, which tend to be closest to the direct beam. But these spots are also the least sensitive to angular perturbations. In fact the spots furthest away from the direct beam most accurately determine orientation. Secondly, the presence of a diffraction spot in an image is a much more reliable indicator of orientation than the intensity of the spot, because to accurately calculate intensity we need to accurately know sample and instrument parameters (even obscure stuff in the projector system) which are usually not known. So a fast kinematical simulation will do just fine for orientation mapping; you can even set all template intensities to 1 (if the rel-rod crosses the Ewald sphere), skipping the entire intensity calculation, and get a similar result.
>
> An added benefit of a kinematical simulation is that the output is already a sparse representation, if you do a multislice simulation you are better of doing image comparisons via FFT-based cross correlations (something of this sort is implemented in py4DSTEM).

Huh, that convinces me.  I guess that in this case it isn't probably worth the effort.
 
> > This method is quite flexible as well. For example you can adjust the interplanar spacing to see if your calibration is slightly off. That just requires another correlation dimension to optimize. As long as the difference is small that should be doable.
> 
> I'm not sure I follow the description but I'd be happy to see an example and review a PR. The way I'm seeing it is you have a list of spots in your image (r, θ, intensity) and the same for your templates, but these lists are not necessarily the same length nor are the closest spots in the same order. So I'm not sure how calculating `delta_r` and `delta_t` would work?

Thank you!  I'll do a more complete work up of this.  The lists aren't the same length.  In many ways it is very similar to your method.  Basically you compare each vector you identify to each of the of the N kinematic calculated vectors.  I was a little naive in my first work up which might be why it was confusing.  Basically you need to create a difference matrix and then apply the same method with the double for loop. I just tried to take the inner for loop and do it in a numpy vectorized fashion. 

> > It also opens up this kind of analysis to people without extensive GPU resources
> 
> The template matching thing I implemented should also work on CPU with relatively modest resources. It will just take 20 minutes instead of 2 (depending on the complexity of the phase that needs to be indexed). But I would be happy to see your vision and its performance!

Hmm, I guess I never really looked very closely at the metrics.  I guess it makes sense.  Your method should be fairly comparable in most cases because it is basically operating on reduced strips of the data.  It does kind of bring into question the reason for having a vector based approach.

CSSFrancis, 2023-01-19:
This might not be helpful but this is now a working example:

```python
import numpy as np
import matplotlib.pyplot as plt
from skimage.draw import ellipse
from scipy.ndimage import correlate

def create_profile(theta_positions,
                   disk_r,
                   theta_r, fully_resolved=True):
    """
    Creates a profile in r and theta which are used to create the equivalent
    of a cross_correlation in r and theta between two disks with major axis
    lengths of disk_r and theta_r

    Parameters
    ----------
    theta_positions: int
        The number of theta positions to sample.
    disk_r: int
        The size of the disk radius in r positions
    theta_r: int
        The radius of the disk radius in t positions.
    """
    profile = np.zeros((int(disk_r*6),theta_positions)) # extra padding to stop wrapping in r
    
    rr, cc = ellipse(disk_r,theta_positions/2, disk_r, theta_r)
    profile[rr,cc] =1
    cross_corr = correlate(profile, profile, mode="wrap")
    cross_corr = cross_corr[disk_r*2:disk_r*4]
    cross_corr = np.roll(cross_corr,int(theta_positions/2), axis=1)
    if fully_resolved:
        cross_corr =np.array([np.roll(cross_corr, i, axis=1) for i in range(theta_positions)])
    return cross_corr

def polar_vector_cross_correlate(polar_vectors,
                                 template_vectors,
                                 cross_correlation_profile
                                ):
    """
    Calculate the cross correlation between a set of vectors and a template of vectors. 
    
    Parameters
    ----------
    polar_vectors: array-like
        A array of polar vectors with r and theta columns
    template_vectors: array-like
        The vectors associated with one template with r and theta columns. 
    cross_correlation_profile: array-like
        A 3d array with dimensions (max_theta, max_r, max_theta) which is used as a look up table
        to determine the intensity at some delta theta and delta r.
    
    """
    _, max_r, max_theta = cross_correlation_profile.shape
    half_r = int(max_r/2)
    
    tv_shape = template_vectors.shape 
    # matrix shape  = (#polar_vectors, #template_vectors)
    delta_r_matrix = np.rint(polar_vectors[:,0]-template_vectors[:,0].reshape(tv_shape[:-1]+(1,))).astype(int) 
    delta_t_matrix  = np.rint(polar_vectors[:,1]-template_vectors[:,1].reshape(tv_shape[:-1]+(1,))).astype(int)

    # only calculate the ones in range. 
    in_range = np.abs(delta_r_matrix) < half_r
    
    profiles = cross_correlation_profile[delta_t_matrix[in_range],delta_r_matrix[in_range],]
    return np.sum(profiles, axis=0)

```

Then testing with some random data...

```python
r_templates = np.random.randint(0,50, (10000,10))
t_templates =np.random.randint(0,180, (10000,10))
templates = np.stack((r_templates, t_templates),axis=-1)
pattern = templates[46].copy()
pattern = pattern + np.random.random((10,2))*3-1.5
pattern[:,1]=pattern[:,1]+5

cross_correlations = np.array([polar_vector_cross_correlate(polar_vectors=pattern,
                                         template_vectors=t,
                                         cross_correlation_profile=profile) for t in templates])

np.unravel_index(np.argmax(cross_correlations),cross_correlations.shape) # returns (46,5)
```

Running on a single core matching vs 10,000 templates with 10 vectors each and 10 identified vectors this takes around .2 seconds to complete.  The benefits are that it isn't really dependent on the number of theta positions you are interested in and you don't have a large of a dataset to deal with. 

As far as time savings go.  If both cases are loaded into RAM they are probably comparable.  It does seem like the vector based approach should be able to beat out the polar image approach but the speed increase seems like it is likely a factor of 1-10 rather than a much larger increase.  Additionally, with the cost of finding the peaks it is probably a decent bit slower that the current method.

I can add this in if we think it is useful...

din14970, 2023-01-19:
> multiplies the r position of the diffraction spots by the strip in theta at that r to get the cross correlation. 

Sort of but not entirely. In the CPU implementation it looks like that because it turned out that slicing the polar image was more efficient in the numba kernel than iterating over all the pixels in the column/strip (one r, all $theta$). A key detail is that the strip is constructed from two slices, one before and one after the $theta$ coordinate. The result is that all the selected strips are shifted over by the $theta$ offset of each template diffraction spot (effectively the same as `np.roll` but I tested that was too slow). When each strip is weighted by the template spot intensity and summed, each pixel in the resulting 1D array will correspond directly to the cross correlation of the image and the template shifted over by $\Delta\theta$. Finding the maximum in this array gives us the optimal in-plane angle of the template (through the pixel index and angular sampling), and the result for that template.

In the GPU implementation I do a similar thing but each GPU thread is responsible for calculating the cross correlation of one template at each in-plane angle.

> Your method should be fairly comparable in most cases because it is basically operating on reduced strips of the data. It does kind of bring into question the reason for having a vector based approach.

I think the real value of a vector based approach would be lower sensitivity to distortions in the image e.g. astigmatism in the projector system or incorrect CL calibration, improved robustness, and less necessity for manual tweaking all kinds of parameters to get reasonable results. The current method can be quite sensitive to pixel noise, or it can "miss" a very important diffraction spot due to some image distortion. If one would use a point-distance based approach the pixel coordinates do not exactly have to coincide, and together with symmetry it is also how we would more naturally think of "goodness of fit". Theoretically, representing a diffraction pattern as spot coordinates and intensities is a true, sparse representation of the information in the image with all the noise filtered out. It would make sense to me that comparing this to computed sparse templates somehow is more robust.

> This might not be helpful but this is now a working example:

Interesting approach, I won't pretend I fully understand it at this moment. You also don't use the intensity of the spots, so somehow you are indeed implementing a distance based approach. I'm not sure how the `create_profile` is supposed to be used or how it works though, I get errors or have to wait a really long time and experience a huge memory spike (+12 GB) during the calculation. I supplied (360, 50, 180), which I interpret as increments of 1 degree, but I'm not entirely sure about the other two arguments. How often would this profile matrix need to be calculated?

> Running on a single core matching vs 10,000 templates with 10 vectors each and 10 identified vectors this takes around .2 seconds to complete. The benefits are that it isn't really dependent on the number of theta positions you are interested in and you don't have a large of a dataset to deal with.

The current implementation has a linear time complexity with x = (number of templates) * (average number of spots per template) / (angular increment). I'm not sure what angular increment you used but let's say 1 degree, then x = 1e5. If you check my benchmarks in Fig. 6 in the paper, I would roughly expect a performance of 100-200 ms per pattern on a single core in the current implementation, faster on multiple cores.

It would be interesting to see how your method scales with various input parameter shapes. I'm also pretty sure your method can still be optimized a lot, so I think there is potential!

CSSFrancis, 2023-01-19:
@din14970 thanks for looking at this again! Your thoughts are very helpful!

> > multiplies the r position of the diffraction spots by the strip in theta at that r to get the cross correlation.
> 
> Sort of but not entirely. In the CPU implementation it looks like that because it turned out that slicing the polar image was more efficient in the numba kernel than iterating over all the pixels in the column/strip (one r, all theta). A key detail is that the strip is constructed from two slices, one before and one after the theta coordinate. The result is that all the selected strips are shifted over by the theta offset of each template diffraction spot (effectively the same as `np.roll` but I tested that was too slow). When each strip is weighted by the template spot intensity and summed, each pixel in the resulting 1D array will correspond directly to the cross correlation of the image and the template shifted over by Δθ. Finding the maximum in this array gives us the optimal in-plane angle of the template (through the pixel index and angular sampling), and the result for that template.
> 
> In the GPU implementation I do a similar thing but each GPU thread is responsible for calculating the cross correlation of one template at each in-plane angle.

I think that makes sense, the GPU is good for doing many small tasks at the same time do the linear approach is much worse.

> I think the real value of a vector based approach would be lower sensitivity to distortions in the image e.g. astigmatism in the projector system or incorrect CL calibration, improved robustness, [...]

Yea maybe a better distance metric would be useful. 

> Interesting approach, I won't pretend I fully understand it at this moment. You also don't use the intensity of the spots, so somehow you are indeed implementing a distance based approach. I'm not [...]

The profile matrix is just my way of making a distance penalty that is more physical.  In a perfect template match you would assume that the kernel is a flat disk and your diffraction spot is a flat disk.  Then the cross correlation between the two would be  what I call the profile matrix.  The idea is that it is only calculated once and then the values are read from it.   The memory spike is because you are creating a rather large spots which in turn creates a large matrix. Maybe better parameters should be something like (360, 5, 5) which means that you are sampling 1 degree increments and the correlated spots are 5 pixels in both the theta and r direction. 

It is a weird way of trying to make a distance based metric that is more physical and is comparable with current methods. The idea is that this is equivalent to the following function:

```python
def full_vector_correlation(v1, v2, radius, num_theta=360):
       im1 = np.zeros(shape=(radius*2, num_theta))
       rr,cc = disk(v1[0],v1[1], radius=radius)
       im1[rr,cc]=1

       im2 = np.zeros(shape=(radius*2, num_theta))
       rr,cc = disk(v2[0],v2[1], radius=radius)
       im2[rr,cc]=1

      cc= correlate(im1,im2, axis=1) # correlate along theta direction
      return cc[v1[0]] # return cc at slice at some r. (can use v2[0] and get the same result)
```

> The current implementation has a linear time complexity with x = (number of templates) * (average number of spots per template) / (angular increment). I'm not sure what angular increment you used [...]

So this implementation will also be linear in time complexity but it is not dependent on the angular increment. The only penalty for increasing the angular increment is increasing the size of the profile matrix which makes the indexing slightly slower.

CSSFrancis, 2023-01-19:
Showing the profile a bit better.....

```python

from scipy.ndimage import correlate

def create_diffraction_strip(vector, radial_r, theta_r, theta_positions):
    rr,cc = ellipse(vector[0], vector[1],radial_r, theta_r, )
    strip = np.zeros((radial_r*4, theta_positions))
    strip[rr,cc]=1
    return strip

fig, axs = plt.subplots(1,3, figsize=(15,5),)
strip1 = create_diffraction_strip([6,45],6,7,180)
axs[0].imshow(strip1)

strip2 = create_diffraction_strip([8,90],6,7,180)
axs[1].imshow(strip2)
cc = correlate_theta(strip1, strip2,)
axs[2].imshow(cc)

axs[0].set_title('Radial Slice of Match Vector 1')
axs[1].set_title('Radial Slice of Template Vector 1')
axs[2].set_title('CC of Match and Template Vector 1')

# create profile 

profile = create_profile(180,6,7)

fig, axs2 = plt.subplots(2,1, figsize=(5,7))
axs2[0].plot(profile[:,6+2,45]) # center plus shift
axs2[1].plot(np.sum(cc,axis=0))

axs2[0].set_title('Cross Correlation from Profile')
axs2[1].set_title('Cross Correlation from strips')
```
[image]
[image]

din14970, 2023-01-19:
Here is an idea on using KDTrees which also kind of works in a similar way:

```python
import numpy as np
from scipy.spatial import KDTree
from numba import njit
from collections import Counter

def rotate_pattern(
    pattern,
    angle,
):
    """For rotating a set of points by an angle in degrees"""
    rad = np.deg2rad(angle)
    rotation = np.array([
        [np.cos(rad), np.sin(rad)], 
        [-np.sin(rad), np.cos(rad)],
    ]).astype(np.float32)
    return np.dot(pattern, rotation)

@njit
def rotate_pattern_spiral(
    pattern,
    delta_theta,
):
    """Creates an array of a pattern rotated around 360 degrees at specified increments"""
    number_of_rotations = int(360 / delta_theta)
    result = np.empty((number_of_rotations, *pattern.shape), dtype=np.float32)
    rotated = pattern.astype(np.float32)
    rad = np.deg2rad(delta_theta)
    rotation = np.array([
        [np.cos(rad), np.sin(rad)], 
        [-np.sin(rad), np.cos(rad)],
    ]).astype(np.float32)
    for increment in range(number_of_rotations):
        result[increment, ...] = rotated
        rotated = np.dot(rotated, rotation)
    return result

def correlate(
    pattern,
    templates,
    delta_theta = 1.,
    distance_bound = 0.5,
    workers=1,
):
    rotated_patterns = rotate_pattern_spiral(pattern, delta_theta)
    number_of_spots_pattern = pattern.shape[0]
    number_of_rotations = rotated_patterns.shape[0]
    number_of_templates = templates.shape[0]
    number_of_spots_template = templates.shape[1]
    spot_index_to_template_index = np.arange(0, number_of_templates, 1/number_of_spots_template).astype(np.int32)
    templates = templates.reshape(-1, 2)
    
    tree = KDTree(templates)
    template_spot_indices = tree.query_ball_point(
        rotated_patterns, 
        distance_bound,
        workers=workers,
    ).sum(axis=1)
    
    best_n_matching_spots = 0
    best_match_template = 0
    best_rotation = 0
    
    for rotation_index in range(number_of_rotations):
        spot_indices = template_spot_indices[rotation_index]
        template_indices = spot_index_to_template_index[spot_indices]
        match_template, n_matching_spots = Counter(template_indices).most_common(n=1)[0]
        if n_matching_spots > best_n_matching_spots:
            best_n_matching_spots = n_matching_spots
            best_match_template = match_template
            best_rotation = rotation_index
        
    return best_match_template, 360 - delta_theta * best_rotation
```

The idea is that the best template is the one where the greatest number of spots match within a certain distance of the spots in the real pattern. With your toy example it also works (but I directly use cartesian coordinates):

```python
number_of_templates = 10000
number_of_spots = 10
distance_range = 50
chosen_template = 1300
x_templates = np.random.randint(-distance_range, distance_range, (number_of_templates, number_of_spots))
y_templates =np.random.randint(-distance_range, distance_range, (number_of_templates, number_of_spots))
templates = np.stack((x_templates, y_templates),axis=-1)
angle_rotated = 10
pattern = rotate_pattern(templates[chosen_template].copy(), angle_rotated) + np.random.random((number_of_spots,2))*3-1.5

best_fit_template_index, angle = correlate(pattern, templates, 1, distance_bound=1)
print(best_fit_template_index, angle)
>> 1300 10

# for plotting
# test_template = rotate_pattern(templates[best_fit_template_index].copy(), angle)
```

One could still modify this to also include information on spot intensity or relative spot distance somehow.

With these conditions I hit 50 ms to run `correlate` and no serious memory pressure. I haven't done any extensive testing on how this time changes with the inputs. I don't like that there is still a python loop in the function but couldn't find a more elegant way to speed up this logic.

CSSFrancis, 2023-01-20:
@din14970 This is very cool.  It gives me a couple more ideas of where to go with this. I think in general this is the right idea of where to go.  Basically this changes the problem from a distance based comparison to more of a search problem using a presorted index.  The nice this is that you only ever have to calculate the KDTree once.  Then you can apply the same tree for every probe position. That effectively halves the computation time to around 25 ms. 

> One could still modify this to also include information on spot intensity or relative spot distance somehow.

I think adding the intensity is very easy.  You already have the index so you can use the index to look up the intensity.  For the spot distance I'm pretty sure that needs to be calculated in a brute force fashion.  Correct me if I am wrong, but my understanding of the KDTree is that while it can determine the nearest neighbors you still need to calculate the distance between the nearest neighbors using a distance matrix or something like that.
 
> With these conditions I hit 50 ms to run `correlate` and no serious memory pressure. I haven't done any extensive testing on how this time changes with the inputs. I don't like that there is still a python loop in the function but couldn't find a more elegant way to speed up this logic.

I'm not sure about the python loop either.  I've been thinking about this problem a fair bit and maybe there is a different way to do this? It seems like the best way to get rid of the loop is to use polar coordinates and directly calculate the displacement in theta.

It seems like maybe using sparse representations/presorting the templates in some way might work?  I have some more time to play around with this this weekend and I can try some other things.

din14970, 2023-01-20:
> Basically this changes the problem from a distance based comparison to more of a search problem using a presorted index. 

I'm not entirely sure how the KDTree works under the hood to be honest, but indeed it's some kind of search index that works based on distance (you can specify what the Minkowski order should be, by default it's 2). Using the method `query` returns both distances and the `n` nearest spots. That method did not work very well since some spots in the pattern have a lot of template spots in direct proximity, and some have none. So basically choosing `n` to get the best answer is difficult, and selecting n = number of points that make up the tree is no better than calculating a distance matrix.

`query_ball_point` is better for this case, as instead of providing the `n` nearest neighbors that should be queried, you provide a search radius and all points inside this radius are returned. The downside is that each input point yields a list of indices. These lists are arranged in an object array. Also no distances are returned so they have to be calculated manually.

What you gain with the KDTree is that you only have to calculate the distance between a tiny subset of the possible points.

> The nice thing is that you only ever have to calculate the KDTree once. Then you can apply the same tree for every probe position. 

Yes

> That effectively halves the computation time to around 25 ms.

Did you time this? I thought the construction of the KDTree didn't take much time at all since data is not copied.

> Correct me if I am wrong, but my understanding of the KDTree is that while it can determine the nearest neighbors you still need to calculate the distance between the nearest neighbors using a distance matrix or something like that.

Yes, but again because of the search radius you have cut down the number of points to calculate the distance to from 100000 for each point to maybe 20 for each point.

> I'm not sure about the python loop either. I've been thinking about this problem a fair bit and maybe there is a different way to do this? It seems like the best way to get rid of the loop is to use polar coordinates and directly calculate the displacement in theta.

The main issue is that `query_ball_point` returns an array of lists, which is difficult to operate on in a vectorized way. What one could do is use `query_ball_point` once to figure out how many points are typically in the search radius (length of the lists in the array), take a safe margin, then use `query` with that value. With `query` you get a pure array, and you can also set a limiting radius. Once you have a pure array it's much more easy to do these kinds of operations in the loop in a vectorized way or with a numba kernel.

> It seems like maybe using sparse representations/presorting the templates in some way might work? I have some more time to play around with this this weekend and I can try some other things.

I'm afraid the templates are already sparse representations...

CSSFrancis, 2023-01-24:
> I'm not entirely sure how the KDTree works under the hood to be honest, but indeed it's some kind of search index that works based on distance (you can specify what the Minkowski order should be, by [...]

That seems like a problem.  My understanding of the KDTree is that it is binary "grouping" of points based on their location in some space.  It is good at finding nearest neighbors because it can search the tree and only identify points nearby.  I think this is the right idea but maybe not the right method.  Because each of the branches doesn't equally split the data then calculating distance has to be done in a brute force fashion. 

> `query_ball_point` is better for this case, as instead of providing the `n` nearest neighbors that should be queried, you provide a search radius and all points inside this radius are returned. The [...]

The query_ball_point function is _very_ dependent on the size of the distance_bound that you give it and the number of points in the area around the points.  0.5 pixels is probably too small, but increasing the distance to 2 or 3 pixels makes the query function run in 30-40 ms rather than 5 ms. This effectively doubles the calculation time.  Maybe there is a way to speed this up by building the tree in a better fashion.

> Did you time this? I thought the construction of the KDTree didn't take much time at all since data is not copied.

Yea, the data isn't copied but it still has to compute the decision tree even if the data isn't copied. 

> Yes, but again because of the search radius you have cut down the number of points to calculate the distance to from 100000 for each point to maybe 20 for each point.

That definitely reduces the time for calculating the distance matrix but you are still stuck with the for loop.  Calculating the distance matrix is not exactly the most time consuming part. I think most of the time comes from how do you sum the distances for each vector-template pair for all the vectors in the pattern.  That step takes some time...  You can try to create a sparse array and then sum or count the instances and then multiply by the inverse of the distances.

I think that part will by time consuming
 
> > I'm not sure about the python loop either. I've been thinking about this problem a fair bit and maybe there is a different way to do this? It seems like the best way to get rid of the loop is to use polar coordinates and directly calculate the displacement in theta.
> > Basically this changes the problem from a distance based comparison to more of a search problem using a presorted index.

> It seems like maybe using sparse representations/presorting the templates in some way might work? I have some more time to play around with this this weekend and I can try some other things.

I'm afraid the templates are already sparse representations...

Sorry that was very poorly worded.  Here's my idea explained a little more carefully. 

Because we are using the same template over and over again it makes sense for us to arrange the templates in such a way that is "ideal" for what we eventually want to do.  Time sorting and organizing the templates can help speed up the operations by a fair amount. 

The easiest way to think of this is by taking all of the possible positions in a polar array (say 100 radial and 360 azimuthal pixels) and pre-compute the distances between that point and all of the templates.  That is only computing 360,000 distances if we assume each template has 10 vectors on average. This dataset will be large, (num_radial_pos, num_azimuthal_pos, num_templates) in size but sparse.  This helps as we are trying to search/sum/slice.

Then for some vector we can directly slice the dataset to identify the correlation/matching.

For example:

```python
def accelerated_template_sparse(templates,
                         num_r=50,
                         num_t=360,
                         max_r=3,
                         max_t=3,
                         r_factor=1,
                         t_factor=1):
    """ Creates and accelerated sparse representation of the templates for a fast look up.
    
    The function returns a sparse representation of the data to limit the memory usage. 
    The returned template is num_r x (num_t*2) x num_templates in size with each index in r,t 
    related to a coordinate index in some image. 
    
    Slicing the template (i.e. template[15, 30:360+30]) returns 1/(1+delt_r*r_factor)+1/(1+delta_t*t_factor)
    as a modified representation of the distance from some template to the pixel position.
    
    Note
    ----
    This method is faster when applied to larger datasets as it spends a larger ammount of time creating
    an accelerated template at the cost of greatly increasing the speed when matching the templates. 
    
    Additionally the size of the template can become rather large as the number of 
    """
    flat_temp = flatten_templates(templates)
    
    num_templates = np.max(flat_temp[:,2])
    accelerated_templates = np.empty(num_r, dtype=object) # smaller data size...
    for r in range(num_r): # iterate through r 
        dr = np.abs(flat_temp[:,0]-r)
        in_radial_range = dr<max_r # only look at vectors in radial_range
        
        dt = np.abs(flat_temp[in_radial_range,1][:, np.newaxis]-np.arange(0,360)[np.newaxis,:])
        in_t_range = dt<max_t # only look at vectors in theta strip.
        
[... 32 lines cut ...]
```

Then when you initialize a the template for matching:

[... cut ...]
max_rot, max_temp = correlate_sparse(pattern, acc_template, n_match=1)
max_rot, max_temp
>>30.9 ms ± 833 µs per loop (mean ± std. dev. of 7 runs, 10 loops each)
```

```python
%%timeit
max_rot, max_temp = correlate_sparse(pattern, acc_template, n_match=5)
max_rot, max_temp
>>60.5 ms ± 663 µs per loop (mean ± std. dev. of 7 runs, 10 loops each)
```

Maybe this isn't really that much faster but this at least includes some information about the distance.  Just thought I would put this out here as an alternative and potentially something interesting...

CSSFrancis, 2023-01-24:
Do we think that this is worth writing up? The performance gains from the other method are decent but not huge.

I probably can't spend too much more time thinking about this :)  At the very least I think that it gives an additional method to compare/ check against and if you already have found peaks/vectors then it is easy to calculate.

pc494, 2023-01-25:
Giving this a proper read at the weekend is now on my to-do list, will feedback then.

CSSFrancis, 2023-01-31:
Interestingly enough the method that I wrote above is basically the same as an Oct tree (I guess in 2 dimensions that might be called a Quad Tree) It is the same as a KD Tree except every node is the same size.  In our case we are just using a pixel notation.  

There is probably a lot faster/ memory efficient way to implement the same thing if it was written in C++.  I haven't looked too much but there might be a good python implementation out there somewhere.

tinabe, 2025-02-11:
We're looking into vector matching again and is now at the beginning stage where we consider different algorithms and ways of accomplishing this. If any of you have any opinions or would like to discuss with us on this topic, please let me know :) !

CSSFrancis, 2025-02-11:
@tinabe I'd love to have a discussion about this! Maybe we can set up a video chat sometime? 

Some other thoughts on the matter since the last time that this was dicussed.  It seems to me like doing the window-normalized cross-correlation ([Diffraction2D.template_match_disk()](https://pyxem.readthedocs.io/en/stable/reference/generated/pyxem.signals.Diffraction2D.template_match_disk.html#pyxem.signals.Diffraction2D.template_match_disk)) _should_ improve the orientation mapping as it reduces the background/ increases the intensity of higher order diffraction which seems to be important for orientation mapping.  In practice I've mostly been looking at non-precession data and I'm kind of starting to suspect that non-precession orientation mapping isn't really a reasonable/ I've been having a hard time getting anything to work.

The next question is does going from a template matched dataset  --> a vector representation actually change anything?  If ultimately we are looking for peaks above some threshold with some distance of a simulated dataset then what is the benefit of converting to a list of vectors first/ is there a set of image transformations which are functionally equivalent. 

I still think there are still really good reasons to try this with a list of vectors, but it might be primarily related to making the refining step faster and allowing for maybe some more heavy processing on the backend.  It would be worth thinking about what this backend processing is.  Can we try to account for ellipticity, or dynamical diffraction?  What about using nearby diffraction patterns to help inform fitting or refine orientations.

tinabe, 2025-02-12:
> [@tinabe](https://github.com/tinabe) I'd love to have a discussion about this! Maybe we can set up a video chat sometime?
> 
That would be great! Let's plan it by email.

> Some other thoughts on the matter since the last time that this was dicussed. It seems to me like doing the window-normalized cross-correlation [...]
Yes, we are using the 'template_match_disk' function a lot too, often as a pre-processing step for peak finding but also directly as a pre-processing step, and it usually works very well! 

I think non-precessed data only works well for some cases, and there seems to be no single workflow that works for every case.

> The next question is does going from a template matched dataset --> a vector representation actually change anything? If ultimately we are looking for peaks above some threshold with some distance [...]

I think there are some important differences, yes. Also, just having a good alternative is great and could hopefully facilitate discussions on uncertainty, accuracy etc.