pyxem/pyxem #901: Problem with Calibration

Psiche91, 2023-01-25:
Hello, everybody,
I'm working on 4DSTEM data with .dm4 extension.
these data have been measured on a FIB lamella in which I have a layer of Silicon FCC with orientation [111] and a layer of Perovskite.
For this data I have the diffraction pattern scale bar which tells me the length of one hundred pixels.
To set the reciprocal space scale in Angstrom^-1 I indexed the reflections of the Si with another software (thus I find out the distance between the center of the pattern and a reflection) and I wrote the following code:

`%matplotlib qt5`
`import hyperspy.api as hs`
`import pyxem as py`
`import numpy as np`
`import pylab as plt`
`import math`

`s = hs.load('./Data/*.dm4')`
`s.set_signal_type('electron_diffraction')`

To take the coordinates of the center and the reflection I know:
`click_coord = plt.ginput(2, show_clicks=True)`
`print(click_coord)`

d is the number of pixel between the known reflex and the center of pattern 
`d=math.sqrt((click_coord[0][0]-click_coord[1][0])**2+(click_coord[0][1]-click_coord[1][1])**2)`
`print(d)`

(0.5308 is the distance between the index reflex in the Si and the centerof the pattern diffraction)
`scale = 0.5208/d ` 
`s.set_diffraction_calibration(scale)`

once done, the scale bar in the pattern changes, the scale bar measures the declared value (0.2A^-1) but if I measure the distance between the known reflection and the center of the pattern I get a different distance than what I expect. 

one more thing, what is the command to see the calibration/scale of `s`? before and after?
because I can't find it in the metadata

I hope I was clear, and I ask you if I'm wrong, or I don't know the right commands to set the calibration.

if you need any other information, just ask.
Have a good day
And thanks a lot

pc494, 2023-01-25:
Hi @Psiche91,

I hope you don't mind, I've edited your issue slightly to be a little easier to parse. 

Calibration is often tricky, but what you are doing looks about right. I've never worked with `plt.ginput` but it's possible that that is returning in unhelpful units. Before I dive deeper, is the `0.5308` to `0.5208` typo only in GitHub or in the code too? That would be a simple explanation.

If I remember correctly `s.axes_manager` is the place to find calibrations.

Psiche91, 2023-02-06:
Hello @pc494 and @CSSFrancis, sorry for delay.
I and @littlelyon we managed to figure out the problem for image scaling.
The collected diffraction data is in 128x128 pixel format, but matplotlib displays them at 512x512.
so we simply divided the distance from the reflection to the center or reflected-reflection distance by a factor of four:

`get_coordinate = plt.ginput(2, show_clicks=True)`
`print(get_coordinate)`

`distance_refl_refl = math.sqrt((get_coordinate[0][0]-get_coordinate[1][0])**2+(get_coordinate[0][1]-get_coordinate[1][1])**2)`
`distance_refl_refl=distance_refl_refl/4.0`

`n_pixel=distance_refl_refl/2.0`
`scale_factor = 0.31894671/n_pixel`
`s.set_diffraction_calibration(scale_factor, center=None)`

Why this division is necessary is obscure to us. Do you have any idea ?

P.S. we used the distance between two reflections symmetrical about the center to reduce the distance error.

Thanks a lot

CSSFrancis, 2023-02-20:
@Psiche91  Sorry I never got back to you!  This slipped to the bottom of my ToDo list and I forgot about responding!

> Hello @pc494 and @CSSFrancis, sorry for delay. I and @LittleLyon we managed to figure out the problem for image scaling. The collected diffraction data is in 128x128 pixel format, but matplotlib [...]

That seems like an effect of [Matplotlib anti-aliasing ](https://matplotlib.org/stable/gallery/images_contours_and_fields/image_antialiasing.html) I've run into this a couple of times and it can be quite frustrating.

Basically matplotlib will sometimes interpolate to make images look a little nicer and that might effect how the points are actually picked.  Unfortunately this setting changes slightly from backend to backend as well as between versions of `matplotlib`

You can try `signal.plot(interpolation=None)`  and that might solve the problem.  This might be something to change upstream in hyperspy.  For scientific imaging I would rather have the interpolation off by default.

closed 2024-01-29 by pc494