pyxem/pyxem #902: Presenting at M&M 2023

CSSFrancis, 2023-01-26:
I was going to present our recent work on pyxem at M&M 2023 in Minneapolis.  There is a session [A13 - Computational Advances in Electron Microscopy](https://www.microscopy.org/MandM/2023/program/descriptions.cfm) which I think would be a good fit for pyxem.  The presentation are rather short but I think it would be a good place to advertise a 1.0.0 version. 

I raised this issue to see about a couple of things:

- Is any one else planning to attend?  We could easily split the content of pyxem and go more in depth into particular features. 
- How do we want to handle co-authors on the presentation. It is a little bit harder to get everyone's eyes on a presentation before presenting it.  I was thinking that if people are interested in being credited for the presentation they should opt in?  I was planning to look at the Git Blame for each part of the talk to give credit to the original author of the code regardless. 
- Is there any research that people would like for me to talk about in addition to giving a general `pyxem` overview?  If people are willing to submit abstracts/ presentations but are not able to come I can present that information. 

@din14970 @pc494 @magnunor @hakonanes

pc494, 2023-01-26:
1) I will ask the Camb lot if they're sending anyone.
2) I'm happy for presentation to go under the name of the presenter with a fair acknowledgement in the final slide of the wider community
3) I'm a pass on new research.

din14970, 2023-01-26:
1) I would have liked to attend an M&M but never got the chance! As I am no longer doing EM in my job and EM related software dev is only an infrequent hobby at this point I can't justify going anymore (plus I'd have to pay it out of pocket).
2) Academic credentials like my name on a presentation is not really that valuable to me anymore, but I appreciate the offer. My honest opinion is that the one who makes the presentation and presents it should be the only name on the title slide, but some advisers may disagree. I'm fine with being in acknowledgements. I do think you should mention that Pyxem was built by people from the EM community all over the world without any centralized coordinating body. I think that's pretty unique, especially compared to some other packages that are developed within one research group. I'd also be interested to hear the feedback you get.
3) Currently I'm not really doing any new pyxem work, but I am working a bit on diffsims (progress is slow, may take quite a while, and it's not research worthy). I'm also working a bit on my [pymatchseries package](https://github.com/din14970/pyMatchSeries/pull/8) which might be interesting to share with the EM community at some point. But again, it might take a while before that is ready.

hakonanes, 2023-01-30:
Thanks for bringing this up, @CSSFrancis.

1. I don't know more than that there are a couple people from the TEM group at NTNU who will attend, excluding myself.
2. I agree with @pc494 and @din14970.
3. You could perhaps briefly (!) mention the pyxem family of packages?

> I am working a bit on diffsims

@din14970, exciting! Is this in any way related to the `ReciprocalLatticeVector` class? If so, it would be good to hear what this is (in a diffsims issue or via email), as this class is very important for kikuchipy's functionality.

din14970, 2023-02-01:
@hakonanes It may touch on that class yes but I would always aim for backwards compatibility. I mainly wanted to improve the speed of simulating the template libraries, and extending the capabilities to enable real projection geometries.

magnunor, 2023-02-09:
1. I'm going to M&M for the `A13 - Computational Advances in Electron Microscopy` session. I'll be presenting some work on structural phase changes, utilizing some of the features from the old `pixstem` library.
2. I think it is ok the way the others suggested: the presenter is the author, with the other contributors in the acknowledgement in the presentation.

randih67, 2023-02-12:
Hi,  FYI -  I am going to M&M and have two (connected)  invited talks in A04 - The Praxis of 4D-STEM - Extracting Information from Biological and Functional Materials and X32 - Technologists' Forum Symposium – 4D STEM Tips and Techniques [Partnering with A04] - I planned to include only authors from NTNU/Trondheim, and give a broad presentation on what we do on aluminium alloys (A04) and more on exp setup and analysis in general (X32) - in both presentations I will tell about and acknowledge Pyxem...  (From Randi Holmestad)

CSSFrancis, 2023-02-12:
Sounds great everyone!  For those of you going to M&M (@magnunor  and @randih67) it would be great to meet up in person, we can come back to that a little later though!  

I'll submit a general pyxem paper/ presentation with a focus on design, scalability and performance scaling form desktop to multi-node distributed computing for a wide range of general 4-D STEM workflows.  Then for the presentation I'll just include a slide advertising the different talks at M&M.