pyxem/pyxem #919: errors in install

xtli7687, 2023-04-07:
Dear developers,

I'd like to express my gratitude for developing such a useful package. 
However, while trying to install the package using "python setup.py develop", I encountered the following errors.

error: Download error for https://files.pythonhosted.org/packages/35/a8/365059bbcd4572cbc41de17fd5b682be5868b218c3c5479071865cab9078/entrypoints-0.4-py3-none-any.whl#sha256=f174b5ff827504fd3cd97cc3f8649f3693f51538c7e4bdf3ef002c8429d42f9f: _ssl.c:980: The handshake operation timed out

Could you please guide me on how to resolve this issue?

Thanks.
Xingting

hakonanes, 2023-04-07:
Hi @xtli7687,

thank you for your considerate words.

Your link points to the Python 3 wheel of the package entrypoints v0.4, i.e. unrelated to pyxem. Did you try a second time? I can install and load pyxem v0.15 without issue on Ubuntu 22.04.

hakonanes, 2023-04-07:
Also, I'm not familiar with the command `python setup.py develop`, do you mean `pip install --editable .[dev]`? Version 0.15 was just released, so you get all the latest updates if you install from PyPI with `pip install -U pyxem`.

CSSFrancis, 2023-04-07:
@xtli7687 @hakonanes I think that `python setup.py develop` was an older way to install a package in a developer mode. It tends to fail sometimes (see https://stackoverflow.com/questions/19048732/python-setup-py-develop-vs-install)

If you want to develop and change the code for the database you can do as @hakonanes says:

Just note that you have to be in the pyxem folder for his code to work. The . above refers to the current working directory so you could also do:

`pip install --editable ~path/to/local/pyxem/[dev]`

closed 2023-04-10 by pc494