pyxem/pyxem #922: Non-Ridgid Registration of 5-D STEM data

CSSFrancis, 2023-04-14:
@din14970 This might be of some interest to you and we could potential integrate with your [pyMatchSeries](https://github.com/din14970/pyMatchSeries)

**Is your feature request related to a problem? Please describe.**

Current detectors are now fast enough that it is possible to acquire multiple 4-D STEM datasets in times that are comparable to traditional STEM.  With a camera running at 50,000 frames per second a 512x512 probe position experiment can be acquired in 5 seconds.  

Multiple repeated acquisitions often have some drift which can be corrected using some type of Non rigid transformation. We have tried this a couple of times and see fairly good results.

**Describe the solution you'd like**

The general process is to:

1. Collect a 5-D STEM datasets (repeated 4-D STEM data) 
2. Calculate a series of Virtual dark field images
3. Perform Nonridgid registration on the VDF images
4. Interpolate the 5-D Datacube using the shifts identified using the NRR.

Of these steps, Number 4 is the hardest because often times you run into pretty substantial RAM requirements.  The way around this would be to Transpose the dataset and then save the data chunked in a different way.  You could load the entire 4-D STEM dataset into RAM but each dataset is usually on the order of 10-20 Gb which limits the mutithreading you can do unless you have extremely large memory per core (~50 GB).  

The [rechunker](https://rechunker.readthedocs.io/en/latest/index.html# ) application is quite good at this but I have yet to figure out a way to integrate it into hyperspy/pyxem naturally.

din14970, 2023-04-14:
Looks interesting but very computationally intensive I would imagine. Not sure about the RAM issue you describe though. For 4., would it make sense to identify the coordinates that each scan position should map to (the result of 3.), then perform bi-linear interpolation on a pixel-wise basis between the 4 nearest diffraction patterns?

CSSFrancis, 2023-04-14:
> Looks interesting but very computationally intensive I would imagine. Not sure about the RAM issue you describe though. For 4., would it make sense to identify the coordinates that each scan [...]

Sorry, I looked back at what I wrote and realized that I didn't exactly say what I meant.  The RAM issue only comes into play if you leave the data chunked in the real space dimension.  This is how we typically deal with data but in this case is extremely suboptimal.

Yes you are correct.  The bi-linear interpolation on a pixel-wise basis is the ideal way to go. That requires that you have the dataset chunked in such a way that the real space dimensions are spanned by one chunk and the reciprocal space dimensions are chunked reasonably.  

In hyperspy that signal might look like `< kx, ky, t |x ,y>  where x and y are your signal axes and kx/ky and t are your navigation axes.

Then we could easily call the `map` function:

`s.map(interpolate,  shifts) # shifts is a signal with dimensions == <kx,ky,t|2> `

Most datasets that people work with are normally chunked fully in the reciprocal space dimension so getting them to be chunked in the real space dimension is slightly difficult especially when the datasets are larger than the available RAM. That is where the Rechunker function comes in.