pyxem/pyxem #923: Zenodo has not updated to 0.15.0

pc494, 2023-04-15:
Raised elsewhere, but still an issue.

              I think the Zenodo might have failed. I suggest we open that as a separate issue and close this.

_Originally posted by @pc494 in https://github.com/pyxem/pyxem/issues/891#issuecomment-1501885189_

hakonanes, 2023-04-16:
99.9% sure the error is this trailing comma:

https://github.com/pyxem/pyxem/blob/d8c3cbd566dc484e496c4d601cce71fe2842b581/.zenodo.json#L66-L70

hakonanes, 2023-04-16:
We could make a 0.15.0.post0 release with this comma (invalid JSON) removed, just like we did for orix 0.9.0.post0.

hakonanes, 2023-04-16:
Even better, I suggest to release a patch 0.15.1 with #921 and the Zenodo file fixed.

pc494, 2023-04-16:
I'll fix this

hakonanes, 2023-04-16:
Great!

I've recently experienced that my editor, PyCharm, help with spotting errors in JSON files (typically these trailing commas).

closed 2023-05-12 by CSSFrancis