pyxem/pyxem #927: R

CSSFrancis, 2023-05-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is, ideally a linked issue.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

pc494, 2023-05-06:
Sounds good to me, ping me for a review.

ericpre, 2023-05-10:
0.15.1 has been tagged but it hasn't been uploaded to pypi. Is it expected?

CSSFrancis, 2023-05-11:
@ericpre yea I think I messed up the 0.15.1 release. It uploaded to Zenodo but publishing to pypi failed cause I didn't increment the version number. I deleted the published 0.15.1 but realize that was probably not the best thing to do.

Still trying to get the hang of some of this stuff.

It might be best to just release a 0.15.2 version after #930 is merged. That being said if you have any advice I'm willing to hear it.

ericpre, 2023-05-11:
An alternative is to delete the 0.15.1 tag, fix the version and push another tag - it may create another zenodo entry, which is possibly not a big deal.

I am about to make a hyperspy-bundle release and I was holding it to include pyxem 0.15.1, but if you think that you will more time, I may make one now and another later after a new version of pyxem is available on pypi/conda-forge.

CSSFrancis, 2023-05-11:
@ericpre I can do that this morning if you like.  I was hoping to merge #930 but can also make the release without it.

CSSFrancis, 2023-05-12:
So it failed to publish to pypi. I think that was just a weird fluke related to pypi experiencing a small outage but I'm not sure. 

It does seem like after this we should think about making our GitHub release action a bit more bulletproof.

ericpre, 2023-05-12:
Indeed, this is bad luck with the timing: https://status.python.org/

Can you try to re-run (from the github web interface) the workflow to upload to pypi?

CSSFrancis, 2023-05-12:
Well the award for the most convoluted release goes to 0.15.1 but we should be good now.

closed 2023-05-12 by CSSFrancis

hakonanes, 2023-05-12:
A release guide would be good!

hakonanes, 2023-05-12:
Good job, btw.

hakonanes, 2023-05-13:
@ericpre @CSSFrancis pyxem 0.15.1 is now available from conda-forge.

hakonanes, 2023-05-13:
I've added the mentioned release guide to PR #931.