pyxem/pyxem #928: Beam cententering issue

PVacek, 2023-05-05:
Dear Pyxem community,
I installed pyxem version 0.15.0 (with hyperspy 1.7.4) to try the new center_of_mass option in `center_direct_beam` and it works well. Sadly, the 3 original methods (blur, interpolate, and cross correlation) started to center to the bottom right corner of the diffraction pattern after this update. Did anyone observe similar behaviour? The `get_direct_beam_position` function already gives wrong positions for the direct beam, but I was not able the find the reason for this.

CSSFrancis, 2023-05-05:
Hmm I haven't had any problems using it recently.  Would you mind sharing your code for this?  I can see if I can duplicate this bug.

What do you mean by the `get_direct_beam_position` gives the wrong position for the direct beam?

CSSFrancis, 2023-05-05:
There are a couple of things that we could consider changing and that might be causing this bug

1. The Center direct beam function occurs inplace which I don't love.  This is a by product of how hyperspy's `map` function works and it defaults to inplace.  Sometimes this leads to weird behavior when you try to center multiple times.
2. It is possible that the other methods don't work because there is some thing they are centering on in that corner.  A simple solution for this is to pass the kwarg `signal_slice` not only does it speed things up quite a bit but it generally gives much better results.

```python
import hyperspy.api as hs
import numpy as np
from skimage.draw import disk

data = np.random.random((5,5,100,100))
x = np.random.randint(40,60, (5,5))
y = np.random.randint(40,60, (5,5))
for ind in np.ndindex((5,5)):
    xx,yy = disk(center=(x[ind],y[ind]), radius=6, shape=(100,100))
    data[ind][xx,yy] = data[ind][xx,yy]+6
s = hs.signals.Signal2D(data)
s.set_signal_type("electron_diffraction")
shifts = s.get_direct_beam_position(method="blur", sigma=4)
hs.plot.plot_images(shifts.T)
hs.plot.plot_images(hs.signals.Signal2D([50-y,50-x]))
```
I tried the code above and got so it seems to be behaving correctly.

[image]
[image]

PVacek, 2023-05-05:
Thank you for the quick reply, I will look at it on Tuesday next week. Have a lovely weekned.

PVacek, 2023-05-09:
If I use `get_direct_beam_position`, it will return the beam position in the bottom right corner of the diffraction pattern instead of near the center of the diffraction pattern where it should be. I attached my centering script, in order for it to work you just need to load your signal as `dp` and then you can select data subset (every x pixels to optimize centering parameters and then run it for the whole scan). I also added my anaconda environment that I am using. 

[env_and_centering_script.zip](https://github.com/pyxem/pyxem/files/11429100/env_and_centering_script.zip)

CSSFrancis, 2023-05-09:
@PVacek Hmm interesting bug.  The problem is that the `half_square_width` is being used to create the signal slice and as a result it is trying to center on a weird part of the pattern because your offset is set to the center of the diffraction pattern.

This part of the code needs to be smarter:
https://github.com/pyxem/pyxem/blob/ee2636723dcad88c9d1cb4c5361568dcbafe2738/pyxem/signals/diffraction2d.py#L827-L835

I'll push a change later tonight and we can release a 0.15.1 version tomorrow to fix this.

In the mean time this works:

```python
pos = dp.get_direct_beam_position(method='blur',
                                  sigma=3,
                                  signal_slice=(-10,10,-10,10)
                                 ) # slice from -10 to 10 in x,y
centered = dp.center_direct_beam(shifts=pos, inplace=False)
```

It also works if you use:

```
shifts_dp ,centered = dp.center_direct_beam(method='blur',
                                           sigma=3,
                                           return_shifts=True,
                                           signal_slice=(-10,10,-10,10),
                                           inplace=False) # Add inplace=False 
```

A couple of suggestions about your code just to maybe offer some suggestions. 

1. You can pass the `inplace=False` parameter to stop you from having to use deepcopy on everything. Almost all methods in pyxem take that kwarg and it will stop things from happening inplace.

2. If you are using a lot of RAM, operating lazily is a good change to your workflow. I rarely use anything else.  It helps with parallelism, (the ram issue above) and is good for scaling workflows.

closed 2023-05-12 by CSSFrancis