pyxem/pyxem #942: Adding a 2nd paper to the README?

pc494, 2023-07-14:
Hello all,

Long time no post. I have finally been able to publish some of my (and others) work using pyxem to understand parameter choices in strain mapping. It's available on arxiv and comes with both data and notebooks (via Zenodo). If I made a suitable PR would people be willing for me to include it in our README?  

**Two-Dimensional Strain Mapping with Scanning Precession Electron Diffraction: An Investigation into Data Analysis Routines**
https://arxiv.org/abs/2307.01071

CSSFrancis, 2023-07-14:
@pc494 That sounds like an excellent idea!  In general we should probably be better at listing the papers which contribute to methods in pyxem. I glanced at the paper and lots of it would be good for redoing the strain mapping tutorial.

I think in an ideal world each example notebook would have an associated paper which describes things more in detail like yours.

hakonanes, 2023-07-14:
I quickly read through the paper last month and found it very readable and well put together. I suggest to put it in the README as it describes a core functionality of pyxem just as the orientation mapping paper does.

pc494, 2023-07-14:
Thanks both, I'll try and get that done over the weekend :)

closed 2023-07-21 by hakonanes