pyxem/pyxem #948: Error in strain calculation

PVacek, 2023-08-18:
Dear Pyxem community,
I recently wanted to create this workflow:
Generate strain distribution -> simulate diffraction pattrens -> create 4D-STEM dataset -> do strain mapping -> compare the results with original strain distrubution to see if everything works correctly
Along the way, me and @JoonatanL discovered an error in how pyxem calculates normal strain (only eps_xx, and eps_yy). See the example with 50% tensile strain below (it is a very large strain, but I will use it to demonstrate the error). The error increases with increasing strain, so it is reasonably low with low strains, but it can be fixed easily.
And as always, let me know if there is a mistake in my code or thought process.

Lets assume that we have a 50% tensile strain,
we have a set of unstrained vectors (DiffractionVectors class)
Vu = [[ 1.,  0. ],   [ 0. , -1.]]
Strained vectors with 50% tensile strain should be:
1.5, but we have the distance in reciprocal space, so the distance is 1/1.5, our strained vectors are then,
Vs = [[ 1/1.5,  0. ],   [ 0. , -1/1.5]]
now I will put relevant code bits from pyxem functions to calculate the strain,
#### from get_single_DisplacementGradientTensor
L, residuals, rank, s = np.linalg.lstsq(Vu, Vs, rcond=-1) 
D = np.eye(3)
D[0:2, 0:2] = L
#### from _polar_decomposition, in this case we have only normal strain, so this will not change anything, but it is a part of the workflow
U = np.array(scipy.linalg.polar(Vs, side='right'))[1]
#### from get_strain_maps
e11 = -U[0,0].T +1
e22 = -U[1,1].T +1
this will give strain value of 0.33 which is obviously not 0.5, in order to fix this we need to change it to:
e11 = 1/U[0,0] -1
e22 = 1/U[1,1] -1
and now we will get the correct value of 0.5.

We therefore suggest to change the code in get_strain_maps to:
```
def get_strain_maps(self):
        """Obtain strain maps from the displacement gradient tensor at each
        navigation position in the small strain approximation.

        Returns
        -------
        strain_results : BaseSignal
            Signal of shape < 4 | , > , navigation order is e11,e22,e12,theta
        """
        R, U = self.polar_decomposition()

        e11 = 1. / U.isig[0, 0].T - 1
        e12 = U.isig[0, 1].T
        e21 = U.isig[1, 0].T
        e22 = 1 / U.isig[1, 1].T - 1
        theta = R.map(_get_rotation_angle, inplace=False)
        theta = theta.transpose(2)
        strain_results = stack([e11, e22, e12, theta])

        return StrainMap(strain_results)
```

Code to reproduce the issue:
```
import numpy as np
from scipy.linalg import polar
Vu = np.array([[1.,0],[0.,-1]])
Vs = np.array([[1/1.5, 0.],[0.,-1/1.5]])
L, residuals, rank, s = np.linalg.lstsq(Vu, Vs, rcond=-1) 
D = np.eye(3)
D[0:2, 0:2] = L
U = np.array(polar(Vs, side='right'))[1]
e11 = -U[0,0].T +1
e22 = -U[1,1].T +1
e11_new = 1/U[0,0] -1
e22_new = 1/U[1,1] -1
print('Current pyxem strain values:',e11, e22)
print('Correct strain values:',e11_new, e22_new)
```

CSSFrancis, 2023-08-22:
@PVacek thanks for the detailed work here!

We should think about releasing a bug fix fairly quickly.

Do you want to create a PR or would you like me to do it? I can walk you through the steps for making a new pull request and then you can be listed as a contributor :)

PVacek, 2023-08-24:
Hi, I am happy to learn how to do the pull request myself. I am currently on holiday in Japan (before the IMC conference), so I'll have time to do it probably in the beggining of October, if that is ok.

closed 2024-01-26 by pc494