pyxem/pyxem #961: `TestMake4dPeakArrayTestData.test_rotation` failure

ericpre, 2023-11-05:
From https://github.com/ericpre/hyperspy-bundle/actions/runs/6760600920/job/18374570541

No idea why it is not caught on CI but fails in the bundle.

```python
================================== FAILURES ===================================
__________________ TestMake4dPeakArrayTestData.test_rotation __________________

self = <pyxem.tests.dummy_data.test_make_diffraction_test_data.TestMake4dPeakArrayTestData object at 0x0000027B71ADB790>

    def test_rotation(self):
        rot0, rot1 = np.zeros((2, 3)), np.ones((2, 3)) * np.pi / 2
        xf, yf = np.ones((2, 3)), np.ones((2, 3))
        semi0, semi1 = np.ones((2, 3)) * 5, np.ones((2, 3))
        pa0 = mdtd._make_4d_peak_array_test_data(xf, yf, semi0, semi1, rot0, nt=1000)
        pa1 = mdtd._make_4d_peak_array_test_data(xf, yf, semi0, semi1, rot1, nt=1000)
>       assert not np.array_equal(pa0, pa1)

D:\a\hyperspy-bundle\hyperspy-bundle\WPy64-31150\python-3.11.5.amd64\Lib\site-packages\pyxem\tests\dummy_data\test_make_diffraction_test_data.py:795: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

a1 = array([[array([[1.        , 1.10102051],
               [1.00628314, 1.10092[182](https://github.com/ericpre/hyperspy-bundle/actions/runs/6760600920/job/18374570541#step:17:183)],
               [1.01256604, 1.100625...15156, 1.10013228],
               [0.98743396, 1.10062574],
               [0.99371686, 1.10092182]])]], dtype=object)
a2 = array([[array([[1.10102051, 1.        ],
               [1.10092182, 0.99371686],
               [1.10062574, 0.987433...13228, 1.0[188](https://github.com/ericpre/hyperspy-bundle/actions/runs/6760600920/job/18374570541#step:17:189)4844],
               [1.10062574, 1.01256604],
               [1.1009[218](https://github.com/ericpre/hyperspy-bundle/actions/runs/6760600920/job/18374570541#step:17:219)2, 1.00628314]])]], dtype=object)
equal_nan = False

    @array_function_dispatch(_array_equal_dispatcher)
    def array_equal(a1, a2, equal_nan=False):
        """
        True if two arrays have the same shape and elements, False otherwise.
    
        Parameters
        ----------
        a1, a2 : array_like
            Input arrays.
        equal_nan : bool
            Whether to compare NaN's as equal. If the dtype of a1 and a2 is
            complex, values will be considered equal if either the real or the
            imaginary component of a given value is ``nan``.
    
            .. versionadded:: 1.19.0
    
        Returns
        -------
        b : bool
            Returns True if the arrays are equal.
    
        See Also
        --------
        allclose: Returns True if two arrays are element-wise equal within a
                  tolerance.
        array_equiv: Returns True if input arrays are shape consistent and all
                     elements equal.
    
        Examples
        --------
        >>> np.array_equal([1, 2], [1, 2])
        True
        >>> np.array_equal(np.array([1, 2]), np.array([1, 2]))
        True
        >>> np.array_equal([1, 2], [1, 2, 3])
        False
        >>> np.array_equal([1, 2], [1, 4])
        False
        >>> a = np.array([1, np.nan])
        >>> np.array_equal(a, a)
        False
        >>> np.array_equal(a, a, equal_nan=True)
        True
    
        When ``equal_nan`` is True, complex values with nan components are
        considered equal if either the real *or* the imaginary components are nan.
    
        >>> a = np.array([1 + 1j])
        >>> b = a.copy()
        >>> a.real = np.nan
        >>> b.imag = np.nan
        >>> np.array_equal(a, b, equal_nan=True)
        True
        """
        try:
            a1, a2 = asarray(a1), asarray(a2)
        except Exception:
            return False
        if a1.shape != a2.shape:
            return False
        if not equal_nan:
>           return bool(asarray(a1 == a2).all())
E           ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

D:\a\hyperspy-bundle\hyperspy-bundle\WPy64-31150\python-3.11.5.amd64\Lib\site-packages\numpy\core\numeric.py:[243](https://github.com/ericpre/hyperspy-bundle/actions/runs/6760600920/job/18374570541#step:17:244)9: ValueError
```

CSSFrancis, 2023-11-05:
This was fixed by #952 but a patch was never released.

ericpre, 2023-11-05:
OK, thanks!

closed 2023-11-05 by ericpre