pyxem/pyxem #973: Issue tracking for 0.17.0

CSSFrancis, 2023-11-14:
This is going to probably be a bit of a code dump release for me as I've been sitting on quite a bit of code waiting for hyperspy 2.0.0 to be merged in. 😅 I'll try to do this in increments so it's not too overwhelming.  That being said I might need a bit of help reviewing code before merging it.

Vector Changes!
-----------------
- [ ] #969 
- [ ] #972 
- [ ] #938 
- [ ] #967 
- [ ] #950 
- [ ] Clean up sub pixel refinement 
- [ ] Add example for sub pixel refinement
- [ ] #899 

Removal
---------
- [ ] #896 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

hakonanes, 2023-11-14:
Great work, sounds like a plan! You can request a review from me, and then I can decline if I don't have time...

CSSFrancis, 2023-11-14:
@hakonanes That sounds good!  Thanks!  @sivborg  and @SyHuang19 Would either/both of you be willing to do some additional reviewing of code?

sivborg, 2023-11-14:
> @hakonanes That sounds good! Thanks! @sivborg and @SyHuang19 Would either/both of you be willing to do some additional reviewing of code?

Yes, I am available for that! I am just now tidying up a new PR, but I'll have some time for reviewing right after.

ericpre, 2024-01-02:
Is there any remaining PRs which is necessary to support hyperspy 2.0? If not, would it make sense to push these to another release, in order to release a pyxem version which is compatible with hyperspy 2.0?

CSSFrancis, 2024-01-03:
@ericpre let me look at this tomorrow just to make sure that is the case and then make the release.

pc494, 2024-01-03:
Are we anticipating 0.17.0 to be compatible with both the 1.x and the 2.x series of Hyperspy (at least in the short term)?

CSSFrancis, 2024-01-03:
@pc494 0.17.0 would be locked to hyperspy 2.0.0 as pyxem 0.16.0 is locked to hyperspy 1.x.x. I think it's not worth the effort to support both at the same time.

pc494, 2024-01-03:
seems sensbile to me, probably worth just putting that up top on the release notes.

CSSFrancis, 2024-01-04:
@ericpre I looked at things a little closer and it would be really good to get #972 merged in. 

The change with `find_peaks` to pass the axes_manager through means that there are like 3 different ways to set the scale and offset etc and I rather do that change in 1 go rather than in pieces over multiple releases. That PR simplifies things a fair bit.

I can remove the slicing if you think that still needs work!

PR #985 (2024-01-22): Release 0 17 0
---
name: Release for 0.17.0
about: Support for Hyperspy 2.0.0

---

**Checklist**
- [x] Update the version number in release_info.py.
- [x] Update the CHANGELOG.rst file with the date and new version number and a description of the changes.
- [x] Update the list of contributors in both the release_info.py and .zenodo.json files. __ Make sure that the Zenodo file is valid JSON __.
- [ ] Commit the changes and push them to the repository.
- [ ] Create a new tag with the “v” + version number (e.g. “v0.16.0”) and make a new release on GitHub.
- [ ] Wait for the CI server to finish  [... cut]
closed 2024-01-22 by CSSFrancis