pyxem/pyxem #984: Logo to SVG (and maybe redesign?)

CSSFrancis, 2023-12-27:
**Is your feature request related to a problem? Please describe.**
I've been playing around with the pyxem logo and was wondering how people feel about:

1. Recreating the pyxem logo as an svg so that we can scale it to any size
2. Defining some colors as a "Theme" for pyxem. I quite like the current colors of the design and think it would be good to have a theme if people want to create a pyxem presentation etc.
3. Adapting the logo to better explain pyxem.  There is lots of amorphous characterization in pyxem so having a crystal as the logo isn't the best descriptor. 
4. @pc494 I know you are against words in logos.  This might be a good time to make 2 logos, a plain logo and then a banner logo with pyxem?

**Describe the solution you'd like**
I think we should keep the current logo but maybe change it slightly.  I like the second option where there is both a crystal and an amorphous blob next to it.  We could also have two logos, one for crystals and one for amorphous.  Or we could just keep things the same. 

Pyxem SVG (No words):
------------------------
[image]

Pyxem SVG Amorphous+Crystal (No words):
---------------------------------------------
[image]

Pyxem SVG Amorphous (Alt Logo):
-----------------------------------
[image]

pc494, 2023-12-30:
I think we should stick to one logo (and having it as an .svg is obviously a good move). 
Maybe the amorphous + crystal design you've got currently?
We could also make the crystal/amorphous different segments different colours to draw out the contrast.

CSSFrancis, 2023-12-31:
@pc494 I think you are right about the two logos.

Let me play around with some colors etc and see if we can make it more distinct. I don't really want to change things too much as I like the logo and colors quite a bit. 

I was kind of going for a simplified/ abstract version of nano crystals crystallizing as that pulls in the crystalline, amorphous and insitu characterization.

hakonanes, 2023-12-31:
The small amorphous crystal with rounded corners is a good idea! Great that you found a way to add to the logo. The current colors go well together, should find an equally good combination if you decide to change them.

CSSFrancis, 2024-01-16:
I created a banner as well:
[image]

The font is Noto (same font that Rosettasciio and lumispy use.  I think it looks pretty good even though the "amorphous" part is missing.  If only we had an "i" :)

closed 2024-01-25 by pc494

jlaehne, 2024-03-11:
@CSSFrancis after the nice redesign, can you also share the svg file. Would be nice to use it for the table in https://github.com/hyperspy/hyperspy-website/pull/57 as the svg scales better than the png