pyxem/pyxem #987: PyFAI dependency issues

CSSFrancis, 2024-01-18:
**Describe the bug**
Pyfai made a couple of changes with the 2024.01 version which are causing pyxem to fail.  

Usually this would be an easy fix as we try to support older pyfai versions, however one of the changes is related to how they define the version which makes the problem much more difficult :)  https://github.com/silx-kit/pyFAI/pull/2021

A couple of things we can do:

1. Submit a PR to pyfai to have both the old and new `.version` and `._version` supported (and maybe the more pythonic and correct `.__version__` 
2. Set a maximum version until the 1.0.0 release
3. Remove pyfai as a dependency and rewrite some of the caking code.

I wouldn't really like to do 3 however we will probably have to do some development of our own caking code to get things working properly with cupy/cuda.  

**To Reproduce**

Downloading pyfai 2024.x.x and above will cause the azimuthal integration tools to fail 

@ericpre do you have any suggestions that might be useful here?

ericpre, 2024-01-18:
What is the issue with `version`? `pyFAI.version` is working with both version?

CSSFrancis, 2024-01-18:
> What is the issue with `version`? `pyFAI.version` is working with both version?

I think that the .version file was added about 3 months ago.  Previously we had to use the private ._version which is why this became an issue. We should just be able to do something like this to solve the problem.

```
try:
    from pyFAI._version import version as PYFAI_VERSION
except ImportError:
    from pyFAI.version import version as PYFAI_VERSION
```

ericpre, 2024-01-19:
Regarding the `pyFAI.version` issue, I have just checked a few version (the older being 0.21.1) and it works fine using `python -c "import pyFAI; print(pyFAI.version)"`. From looing at pyFAI code, it seems the file from which the version is retrieved is written at installation time, could it be that your install got corrupted when you changed pyFAI version?

CSSFrancis, 2024-01-19:
@ericpre thank you for doing that!  That problem at least should be easy to fix.

The other bugs introduced in 2024.1.0 are a bit harder track.  For instance the "bbox" method doesn't work with signals that don't have any navigation dimensions.  I think that is actually a result of the `blockwise` function trying to operate on an empty numpy array of shape (0,0).  This normally doesn't cause an issue but pyFAI is throwing an `AssertionError` which isn't properly caught by dask.

On top of that there are some weird extra checks for the Ewald sphere diameter which we are falling outside of the limits for causing a .  I think we are starting to really feel some of the effects of using a package that wasn't made with some of the electron based assumptions in mind.  

I can get the tests to pass but I'm not really comfortable with saying that things aren't going to break.

ericpre, 2024-01-20:
If these issue are not straightforward and would need some time to fix, I think that it would be useful to have a release with a pin on pyfai because currently it is broken with pyfai 2024.1, right?

CSSFrancis, 2024-01-20:
@ericpre it should be pinned right now for release. That might be my vote. 

It's up next on my list of things to change. So it shouldn't stay like that

CSSFrancis, 2024-02-03:
@pc494 What are your thoughts on removing pyfai as a dependancy.  I quite like the new azimuthal integration method and feel like pyfai is fairly redundant.   

With the 1.0.0 version I would maybe suggest:

1. Removing pyfai as a dependancy. Maybe we could add an example of how to use it using the map function if you want but it's too much work to continually support. 
2. Move to just 1 method for azimuthal integration. I think too many options causes lots of confusion. We want the intensity to be conserved (it is) and it to be fast (it is). I don't see much reason to add a less accurate method.

pc494, 2024-02-05:
If you're happy with the integrator I'm more than happy for us to drop Pyfai.

CSSFrancis, 2024-02-05:
@pc494 I think I'm happy with it.  The only thing I'd like to double check is that I correctly accounted for the Ewald sphere.  I've never actually see a good description/ picture of how to account for that.  This is how I've been doing it:
[image]

CSSFrancis, 2024-02-05:
I'll clean this up, maybe add some equations and then add this to the documentation.  I feel like this is one of those things that people who have been in the field for a long time just assume people know and it never gets appropriately taught.

CSSFrancis, 2024-02-05:
But there is also this case where the distance you want to measure is actually the cord of the sphere.  Maybe it really doesn't matter.  Maybe this is already accounted for in diffsims.  Maybe I should think less about this and just accept that it works....

[image]

CSSFrancis, 2024-03-26:
@pc494 Nothing about this is blocking and if we want to drop pyfai we can pretty easily do it now.  I've been using this a bit but still need to make more documentation about Ewald sphere correction.

micwoj92, 2025-06-24:
Shouldnt this issue be closed now? No longer depends on pyfai