pyxem/pyxem #991: Issue Tracking for 0.18.0 Release

CSSFrancis, 2024-01-22:
Release 0.18.0 (The Orientation Mapping Overhaul)
======================================

@viljarjf we are finally going to make Orientation Mapping easy! 

My hope is that this is the last release before the 1.0.0 release! There are still some changes/ deprecations that need to be done and then we can really start to purge some of the code.

To Do List:

- [ ] Fix Azimuthal Integration
    - [ ] #987 
    - [ ] #989 
- [ ] Refactor Orientation Mapping (finally)
  - [ ] Make a Class to hold Orientation Mapping Results
  - [ ] Make Orientation mapping a function of Polar2D
  - [ ] Add in #967 
- [ ] Other things that need doing 
 - [ ] #980 
 - [ ] #901 
 - [ ] Fix up Radial/ ellipse tools
 - [ ] Speed up multi VDF creation using the map function
----------------------------------------------------------------
- [ ] Diffsims (Merge https://github.com/pyxem/diffsims/pull/201)
- [ ] Make a 0.6.0 Release

pc494, 2024-01-26:
I am going to use a milestone to track this as well.

CSSFrancis, 2024-02-17:
> #### In the pipe for 1.0.0
> * Return coverage to 100%
> * Set up a functional cron job

@pc494 What exactly do you mean by a functional cron job?

pc494, 2024-02-20:
> > #### In the pipe for 1.0.0
> > 
> > * Return coverage to 100%
> > * Set up a functional cron job
> 
> @pc494 What exactly do you mean by a functional cron job?

as far as I can tell we don't currently run the testing suite at some fixed time interval. In the past (when we still used TravisCI) this used to happen once a week or so. The main purpose is to spot downstream changes so that they can be corrected for when they arise, rather than making an (unrelated) PR at some point in the future fail.

CSSFrancis, 2024-02-28:
> as far as I can tell we don't currently run the testing suite at some fixed time interval. In the past (when we still used TravisCI) this used to happen once a week or so. The main purpose is to [...]

@pc494 ahh makes sense.  I just wasn't thinking...  The tutorials do this at the moment (however they are failing 😬).

CSSFrancis, 2024-02-28:
@pc494 What do you think about trying to push for the 0.18.0 release soon.  What is still needed to be done (besides the large and daunting https://github.com/pyxem/diffsims/pull/201 (and #1018)

pc494, 2024-02-29:
> @pc494 What do you think about trying to push for the 0.18.0 release soon. What is still needed to be done (besides the large and daunting [pyxem/diffsims#201](https://github.com/pyxem/diffsims/pull/201) (and #1018)

I am keen. I am out-of-office until early the middle of next week though. Is a target of March 22nd to far out?

CSSFrancis, 2024-03-01:
@pc494 that sounds good! It might be nice to get this done before the end of March :) these things always take a bit longer than anticipated.

pc494, 2024-03-12:
Status update. I'm grinding along with #996 (see #1030) and am happy to review stuff but that'll be me for this release.

pc494, 2024-03-13:
I think it would be nice to deal with #950 and #967 while before `0.18.0` as well.

pc494, 2024-03-22:
> @pc494 What do you think about trying to push for the 0.18.0 release soon. What is still needed to be done (besides the large and daunting [pyxem/diffsims#201](https://github.com/pyxem/diffsims/pull/201) (and #1018)

If I'm following correctly pyxem/diffsims#201 now needs an orix patch fix? In which case I'm going to focus some attention there as it's probably our blocker now.

CSSFrancis, 2024-03-22:
> If I'm following correctly [pyxem/diffsims#201](https://github.com/pyxem/diffsims/pull/201) now needs an orix patch fix? In which case I'm going to focus some attention there as it's probably our blocker now.

That is correct!

pc494, 2024-06-22:
This made it into the wild

closed 2024-06-22 by pc494