pyxem/pyxem #995: Moving Release Next Major Branch into main

pc494, 2024-01-26:
We currently have 22 outstanding commits in RNM that (I think) should be shifted over to main. Although RNM seemed like a sensible approach at the time, I think going back to a single trunk branch might be best in the short term.

@CSSFrancis do you know what the outstanding RNM commits do? Are they safe for 0.18.0?

CSSFrancis, 2024-01-26:
@pc494 I would just get rid of that branch... It was from a time when we were intending to make side by side 1.0.0 and 2.0.0 releases for Hyperspy and pyxem.  I was very ambitious at the time.

All of those commits are implemented somewhere currently in pyxem so I wouldn't worry!

pc494, 2024-01-27:
Perfect will do that and close this issue.

closed 2024-01-27 by pc494