pyxem/pyxem #996: Reorganisation of utils

pc494, 2024-01-27:
So I got started on the utils folder in #994 and I think I've got a handle on what we have. Broadly `utils` modules fall into three categories.

## Internal use only

e.g. `_deprecated`, `_slicer`

Functionality that the devs need but that we either don't expect or don't want end users to make use of. My plan here is to simply be more explicit with these modules (i.e making the private modules, rather than just private functions). For example I don't think we want our users trying to use our cuda code themselves . So [cuda_utils](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/cuda_utils.py) would be moved to a private module (with required deprecations etc).

## Support for niche functionality

We originally adopted a utils-heavy pattern. Many classes have supporting functions (that generally get shoved in `map`) that have their own util folders. I would consider it better to just move such functionality directly into the module that defines the class. This (for example) is what I've done in #994 for the utils that support the `ReducedIntensity` classes.

## True, user-facing utils

An obvious example of this group is [plotting_utils](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/plotting_utils.py). We want users to use them, they are utils. I don't plan to change much about these except to rename modules to drop the trailing `util'. I think it's uncontentious to suggest that:

```python
from pyxem.utils.plotting_utils import *
```
is weird when you could have

```python
from pyxem.utils.plotting import *
```

with a deprecation warning on the old imports.

pc494, 2024-02-20:
#1020 didn't update the changelog. Will sweep this up when I do the next batch of utils

pc494, 2024-03-12:
### Updated state of play for 0.18.0

#### No change 

[__init__.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/__init__.py)
[_background_subtraction.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/_background_subtraction.py)
[_cluster_tools.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/_cluster_tools.py)
[_deprecated.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/_deprecated.py)
[_slicers.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/_slicers.py)
[ri_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/ri_utils.py)

#### Renames 
	
[_azimuthal_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/_azimuthal_utils.py)
[_subpixel_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/_subpixel_utils.py)
[calibration_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/calibration_utils.py)	

#### More work

[correlation_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/correlation_utils.py)	
[cuda_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/cuda_utils.py)
[dask_tools.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/dask_tools.py)
[expt_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/expt_utils.py)
[indexation_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/indexation_utils.py)
[insitu_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/insitu_utils.py)
[labeled_vector_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/labeled_vector_utils.py)
[lazy_tools.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/lazy_tools.py)
[pdf_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/pdf_utils.py)
[pixelated_stem_tools.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/pixelated_stem_tools.py)
[plotting.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/plotting.py)
[polar_transform_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/polar_transform_utils.py)
[pyfai_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/pyfai_utils.py)
[radial_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/radial_utils.py)
[ransac_ellipse_tools.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/ransac_ellipse_tools.py)

[segment_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/segment_utils.py)
[signal.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/signal.py)
[symmetric_vector_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/symmetric_vector_utils.py)
[vector_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/vector_utils.py)
[virtual_images_utils.py](https://github.com/pyxem/pyxem/blob/main/pyxem/utils/virtual_images_utils.py)

magnunor, 2024-03-14:
I should really look through some of the old `pixstem` ones, as there are probably several which need some tidying.

But I'm not sure exactly when I'll have the time.

pc494, 2024-03-17:
So with #1032, #1033, #1034 and #1035 I've got this down to 8 that need addressing. If they get technical I'll raise separate issues from now on.

CSSFrancis, 2024-03-27:
@pc494 I don't know if we are going to get much farther with this. I'm not sure that the signals should be private.  There are too many instances where we want to directly create a signal that I think they should be public.

pc494, 2024-03-27:
> @pc494 I don't know if we are going to get much farther with this. I'm not sure that the signals should be private. There are too many instances where we want to directly create a signal that I think they should be public.

As in the functions in
https://github.com/pyxem/pyxem/blob/main/pyxem/utils/signal.py ?

CSSFrancis, 2024-03-29:
> As in the functions in https://github.com/pyxem/pyxem/blob/main/pyxem/utils/signal.py ?

I completely forgot we have a utils/signals.  Yes that __Should__ be private.  I was thinking you were trying to make the signals module private and I was a bit confused.

PR #1055 (2024-04-07): Closing out the utils reorganisations for now
---
Closing out the utils reorganisations for now
---

**Checklist**

- [x] implemented
- [x] update the Changelog
- [x] Fix the docs build
- [ ] mark as ready for review

**What does this PR do? Please describe and/or link to an open issue.**
- Marks all code in the util file `radial_utils` as deprecated, closes #1002 
- Moves `signal` code to a private module
- Renames `expt_utils` to simply `diffraction`
- Moves `dummy_data` down into `data`, closes #1049 

In doing so we close #996
closed 2024-04-07 by CSSFrancis