pyxem/pyxem #1002: Deletion Candidate: Radial Utils

CSSFrancis, 2024-01-29:
Looking through the Radial Utils folder I think we can remove this entire folder.  The beam centering based on an ellipse is useful but it would be much batter as a `map` like function.  I have lots of code that does this so I can fairly easily add an example of this.

I would much rather have this as an example and maybe use the determine_ellipse function in `ransac_ellipse` tools instead. The benefit is that the ellipse is returned as an affine transformation which is more in line with the rest of pyxem.

pc494, 2024-01-29:
This seems sensible, but if you could hold off for a couple of days so I can have another look I would appreciate it :)

pc494, 2024-02-06:
As long as it doesn't cause @magnunor problems I'm happy for these to go.

pc494, 2024-03-31:
Plan is to mark these all as deprecated.

PR #1055 (2024-04-07): Closing out the utils reorganisations for now
---
Closing out the utils reorganisations for now
---

**Checklist**

- [x] implemented
- [x] update the Changelog
- [x] Fix the docs build
- [ ] mark as ready for review

**What does this PR do? Please describe and/or link to an open issue.**
- Marks all code in the util file `radial_utils` as deprecated, closes #1002 
- Moves `signal` code to a private module
- Renames `expt_utils` to simply `diffraction`
- Moves `dummy_data` down into `data`, closes #1049 

In doing so we close #996
closed 2024-04-07 by CSSFrancis