pyxem/pyxem #1006: Deletion Candidate: dask_tools

CSSFrancis, 2024-01-29:
I think we can pretty safely remove most if not all of the dask_tools:

Ones that should be kept:

- align_single_frame
- _intensity_peaks_image_single_frame --> make into a `map` function of `DiffractionVectors` signal.  

The rest can be fairly easily removed with I think little to no effect on the functionality.

@magnunor any objections?

magnunor, 2024-01-30:
Yeah, most of it is legacy code which was effectively rolled into HyperSpy's dask-based `map` functionality.

> _intensity_peaks_image_single_frame --> make into a map function of DiffractionVectors signal.

Sounds good!

------

Some of the other functions:

* `_peak_refinement_centre_of_mass` similar functionality in `DiffractionVectors`?
* `_peak_find_log` og `_peak_find_dog` this should be covered by HyperSpy's peak finder?

CSSFrancis, 2024-01-30:
> * `_peak_refinement_centre_of_mass` similar functionality in `DiffractionVectors`?

This should be covered by #980!

CSSFrancis, 2024-02-02:
We should merge #1005  and #980 and then we can go ahead with this!

pc494, 2024-03-22:
Much of this was done in #1040, but I think we could do with another loop around, so will leave this open.

PR #1047 (2024-03-26): Finishing up 1.0.0 adjustments of utils.dask_tools/_dask
---
Finishing up 1.0.0 adjustments of utils.dask_tools/_dask
---

**Checklist**

- [x] implementation steps
- [x] update the Changelog
- [x] mark as ready for review

**What does this PR do? Please describe and/or link to an open issue.**
I've ended up moving everything into `_dask`. It's also all now private. For speed/simplicity, I've left the old implementations with deprecations in dask_tools, but afaik they aren't called by any internal code now and can be deleted on the release of 0.18.0.

close #1006
closed 2024-03-26 by pc494