pyxem/pyxem #1011: Potentially changing our approach to affine transforms

pc494, 2024-02-02:
At the moment `Diffraction2D` provides several methods that perform specific affine transforms in a map like way.

We have `dp.apply_affine_transformation` which takes an affine transform but then also shift/rotate and two flip methods (all of which look like they were originally in pixstem). 

It would be fairly simple to rewrite the internals of shift/rotate/flip to call `apply_affine_transformation` similar to what @CSSFrancis is doing in #1005. However I think I would prefer splitting steps up with a workflow like

```python

from pyxem.utils.affines import get_rotation_affine

z = get_rotation_affine(30, angle_in_degress=False)
dp.apply_affine_transformation(z)
```

This is a bit more modular and would facilitate easier testing. What are our people's thoughts?

CSSFrancis, 2024-02-02:
The better thing might be to add the `flip` method to hyperspy upstream.  

It would be nice to be able to do:
```python
s.flip(axis=1)

# or

np.flip(s, axis=1)
```
and have them do the same thing.

pc494, 2024-02-02:
> So I might actually go the opposite direction and say that I don't love the `apply_affine_transformation` function. It will resample the data which results in lots of pixel splitting/ blurring which is difficult to quantify. The shift and flip methods on the other hand are clean 1:1 pixel transformations.
> 
> It's part of the reason that I've been handling the affine transformation after the fact. For example in the azimuthal integration we apply the affine transformation to the resampling rather than first warping the data. For things like diffraction vectors I would like to apply to transformation to the peaks rather than to the dataset before hand as it's a bit cleaner.

so there is a historical reason for some this. When I was a student a lot of data was coming off of Stingray camera's that induced a (large) geometric distortion that needed to be corrected. This was basically always the first thing we did after loading. If that is no longer the case we may want to revisit some of the ideas.

I would also be a little bit surprised if the skimage implementation can't deal with flip (and translate) correctly. It looks like it sort of has a sense that those are special cases, [signature here](https://scikit-image.org/docs/stable/api/skimage.transform.html#skimage.transform.AffineTransform).

pc494, 2024-02-10:
I think some of this still make sense. Specifically having `.rotate` call `apply_affine_transformation`. I've gone off convincing users to make the matrices themselves.

I think `.flip` can just map the relevant numpy function too.

I think that would just leave shift, which I think has two cases. Case (1) is two integer shifts, which I agree shouldn't resample the data. Case (2) is fractional shifts, at which point I think using the `.affine_transformation` API is fine. 

@CSSFrancis are you happy for me to get started on these?

CSSFrancis, 2024-02-10:
@pc494 yea if you want to make the change then go for it!

pc494, 2024-02-15:
added a checklist into the original post.

pc494, 2024-02-24:
Although I still think these changes might have improved the API I think now might not be the best time. Instead, I think we should deprecate f`lip_x` and `flip_y` in time for `0.18.0` and leave the rest for a later moment (e.g the changes to `.rotate` are non-breaking).