pyxem/pyxem #1013: Diffraction2D.filter should either be removed or refactored to use Signal2D.map

pc494, 2024-02-07:
I think it was the pixstem approach to UDF's but as of 2.0 ,map handles this nicely.

https://github.com/pyxem/pyxem/blob/da7c8f6d730d8cbc3ebbd9c504b7186180ee6c7b/pyxem/signals/diffraction2d.py#L1086

CSSFrancis, 2024-02-07:
@pc494 actually that was a newer addition by me :) The difference between the filter function and the `map` function is that the filter function operates on the entire 4-D Block rather than just the signals.  

I like to use it for gaussian_filtering in the navigation dimensions.  

Here's an example of how it works. 

https://pyxem.readthedocs.io/en/stable/examples/processing/filtering_data.html#sphx-glr-examples-processing-filtering-data-py

pc494, 2024-02-08:
I'm not sure I quite understand what's going on here, but code that people like using doesn't need to be deleted.

closed 2024-02-08 by pc494