pyxem/pyxem #1027: Better Strain Visualization

CSSFrancis, 2024-03-08:
**Is your feature request related to a problem? Please describe.**
@pc494 I've been playing around with some different ways to visualize strain in datasets. 

Something like:

[image]

In this case the little inset box will update as you move around the sample.

I'm not an expert on strain mapping though and it would be good to get some insight into how to best visualize things like Negative strain etc?  Maybe draw a circle and have the arrows point from the surface of the circle either in or out?

Also how should we represent Exy strain?  It would also be good to have a frame of reference for the actual sample in real space if possible. We can rotate the box so that the box aligns with real space and the vectors align with reciprocal space?  I just need a little help trying to wrap my head around some of this....

**Describe the solution you'd like**
@ericpre I also am having trouble with the arrows not liking to use the "axes" transform.  I think this is a side effect of something in the Quiver class that is forcing the coordinate transform.

CSSFrancis, 2024-03-08:
Something like this where the inset is rotate to match the real space x,y coordinate system

[image]

CSSFrancis, 2024-03-08:
I think we should be able to just make this a function of the `StrainMap` class and then add something that saves any rotation applied to the data from `StrainMap.rotate_strain_basis` and then applies that rotation to the inset box?

ericpre, 2024-03-10:
I would keep it simple and plot the vector basis on the strain maps themselves, as in for example:
https://github.com/ericpre/gpa_demo/blob/main/GPA_defocus_serie.ipynb

What are you trying to display on the diffraction pattern? Maybe you can display some markers representative of the reference pattern / lattice overlay on the diffraction pattern to visualise the strain measurement on each pattern, as a way to visually assess the "measurement quality"?

CSSFrancis, 2024-03-11:
> I would keep it simple and plot the vector basis on the strain maps themselves, as in for example: https://github.com/ericpre/gpa_demo/blob/main/GPA_defocus_serie.ipynb

That seems like a good idea.  It would be easy to plot the vector basis on the strain maps.  

> What are you trying to display on the diffraction pattern? 

I think there are lots of times where you see something in the strain map that looks "weird" and it's always a good idea to look at the diffraction patterns and the fit for each point. I thought about something like this:

[image]

Where the arrows are the vectors from the unstrained basis to the vectors at the strained navigation position. The problem is that the displacements are small and so it's a bit hard to see the arrows.  Maybe just plotting circles would make things easier to see. I like the idea of plotting vectors as I think it makes it a bit easier to see small changes which are a bit more difficult to see in the strain maps. 

Here's what plotting the circles looks like where white is the unstrained basis and yellow vectors at some position.

[image]

> Maybe you can display some markers representative of the reference pattern / lattice overlay on the diffraction pattern to visualize the strain measurement on each pattern, as a way to visually assess the "measurement quality"?

I think that the above visualization are kind of crowded which is why I went to the idea of just plotting the vectors but maybe that just makes it more complicated.

ericpre, 2024-03-11:
I suspect that your first example misled me to think you were referring to the vector basis!

When displaying both markers (reference and measured) it should be easy to use various type of marker (for example circles, dots, crosses) with colour coding, etc. You can also make the colour coding of the marker (circle, dots or arrow) depends on the magnitude of the strain (https://hyperspy.org/hyperspy-doc/current/auto_examples/Markers/arrows_navigation.html) and match it to the colour map used in the strain maps.

If the marker(s) are overcrowded, I would expect that there is enough flexibility with the marker to find to way to display nicely. When zooming in, it should also appear less crowded.

pc494, 2024-03-13:
Sorry to be boring but is this a pre or post `0.18.0` project?

CSSFrancis, 2024-03-13:
> Sorry to be boring but is this a pre or post `0.18.0` project?

I just thought it would be nice for the paper :) right now the strain section just says we can do strain mapping and it would be nice to say we can do strain mapping. And here is what it t
Looks like.

pc494, 2024-03-13:
Great, sounds like I can put it in my "worry about post 0.18.0 pile then" :)