pyxem/pyxem #1029: Saving 1D intensity profile after Azimuthal integration - tutorial 7

AZangiabadi, 2024-03-12:
Hi,
Great work on developing pyxem!
This may sound silly. I am still learning how to work with pyxem with your tutorial files.
How can I save my data after Azimuthal integration (tutorial 7, line 169) into a hspy file? I would like to load it into tutorial 8 to get pair distribution function. 

integration  = dp.get_azimuthal_integral1d(npt=100, mask=mask.data)
integration2d = dp.get_azimuthal_integral2d(npt=100, mask=mask.data)

my goal is:
integration.save('1D-pattern.hspy')

Thanks

pc494, 2024-03-12:
Hi there,

Have you run the code you suggest here? If you have, what happened?

CSSFrancis, 2024-03-12:
@AZangiabadi There is ab warning that gets thrown saving azimuthally integrated signals just saying that it can't save the  AzimuthalIntegrator object but the data will save!

CSSFrancis, 2024-03-12:
@AZangiabadi As a work around you can just use:

`integration.metadata.Signal.ai=None`

It used to save fine even if it threw an error.  The problem is related to the AI object changing and not being serializable.  It's at least part of the reason that we are moving away from it.

AZangiabadi, 2024-03-12:
Thank you. That worked!

closed 2024-03-13 by pc494