pyxem/pyxem #1051: Package Authentication and Pushing to Pull Requests

CSSFrancis, 2024-03-26:
**Describe the bug**
I'm  not exactly sure what the best way to handle this is but I've been struggling to push to other peoples branches as a maintainer.  The main issue appears to be something to do with 3rd party authentication 

https://docs.github.com/articles/restricting-access-to-your-organization-s-data/

I think that added security is a good thing but I was curious if people had different ways of getting around this?  I haven't been able to get either the command line or PyCharm to work for this, but maybe I just have the wrong access key associated pushing to GitHub. 

@ericpre Do you have a suggestion for the right way to handle/do this?

CSSFrancis, 2024-03-27:
I requested oauth permissions per

https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-your-membership-in-organizations/requesting-organization-approval-for-oauth-apps

How do we feel about this?  I think this should be fine.

pc494, 2024-03-27:
Has this improved since you were bumped up to maintainer? Being able to push to other branches tends to be an exception rather than a rule on Github in my experience.

pc494, 2024-03-27:
For what it's worth, in https://github.com/pyxem/orix/pull/469 I tried to use the CLI instructions at the bottom to push changes to a contributors branch (which did not work). However taking the pull request down (tiny little button, see image)

[image]

did seem to work...

CSSFrancis, 2024-03-27:
@pc494 (and anyone else who is having problems with this) it seems like you need to make sure that the oauth is allowed for some third party application.  Otherwise I think you need to be very specific about how you are authenticating to make sure that GitHub doesn't block it.

Requisition oauth above seemed to work well!

CSSFrancis, 2024-04-25:
Closed.  This seems to be a 3rd party application verificiation issue that I have fixed now.

closed 2024-04-25 by CSSFrancis