pyxem/pyxem #1070: Making a 0.18.0 Release

CSSFrancis, 2024-05-08:
I think I will proceed with a 0.18.0 release tonight so that pyxem can be on Conda in time for the Diamond workshop.  We can try to do a 0.19.0 release maybe over the weekend or early next week if we can get the Orientation Mapping changes in. 

@pc494 @ericpre @magnunor @hakonanes  any thoughts?

hakonanes, 2024-05-08:
Not too familiar with the state of pyxem at the moment to comment. But, I have no objections to two minor releases back-to-back. Good luck!

CSSFrancis, 2024-05-08:
@hakonanes Thanks! I think @pc494 wanted to make a release a bit ago and I naively thought that #1018 would be quick....

magnunor, 2024-05-11:
> We can try to do a 0.19.0 release maybe over the weekend or early next week

I think this is a bit risky with the Diamond workshop happening next week. While most people will be using the HyperSpy bundle, some will like to install it in their own Anaconda installs. And I just know having a different version in the hyperspy bundle, and on conda-forge will lead to some issues during the workshop.

I would rather do this after the workshop has finished. Or maybe to an early release as a release candidate.

CSSFrancis, 2024-05-11:
@magnunor I think you are probably right as much as I might not want to admit it :)my plan was to make a rc and a colab/ binder.

pc494, 2024-06-22:
We did this!

closed 2024-06-22 by pc494