pyxem/pyxem #1077: 0.19.0rc

CSSFrancis, 2024-05-13:
**Is your feature request related to a problem? Please describe.**
It might be nice to make a RC for 0.19.0 with the orientation mapping that way we can test and make sure that things are good/working the way that we want them to be. 

**Describe the solution you'd like**
- [ ] Merge #1076 
- [ ] Make `to_crystal_map` function 
- [ ] Anything else?

magnunor, 2024-05-14:
When should the `DPCSignal` changes be included? `0.19.0`? Or later?

I will not have time to work on this before next week. So feel free to make an early RC without it.

pc494, 2024-05-25:
> - [ ]  Make `to_crystal_map` function

Did you know about?
[image]

CSSFrancis, 2024-05-28:
> Did you know about? 

@pc494 yea I've been making a new class rather than adding to the exisisting classes mostly because the orientation mapping results are slightly different.  I've now added in the `to_crystal_map` method.

CSSFrancis, 2024-05-29:
So I think that we are quickly approaching the point where we should try to make a `.rc` for 0.19.0.  My plan is to:

- [ ] Merge Outstanding PRs 😬  (Ideally by June 1st 🤞)
    - [ ]  #1076
    - [ ]  #1057 
    - [ ] #1081 
    - [ ] #1080 
- [ ] Update Tutorials with the idea that this is a good secondary check to identify bugs and work through full workflows.  I think @tinabe @magnunor @sivborg @viljarjf @rbjorge etc  that  for the workshop should be able to base any notebooks on the .rc but there shouldn't be major changes for anything other than the Crystal Orientation Mapping and the STEM DPC.
     - [ ] Update Strain Mapping Tutorial https://github.com/pyxem/pyxem-demos/pull/94
     - [ ] Update Orientation Mapping Tutorial (I will make a PR for this but @viljarjf if you have something I would love to add that as well!)
     - [ ] Update ML Notebook @tinabe It might be good to add your notebook for ML to the Pyxem demos!
     - [ ] Add a Large Data Processing Notebook (This one is on me)
- [ ] Make a pyxem 0.19.0 Release -- > June 7th should probably be the last day for trying to do this, but that gives us time for it to get up on Anaconda.  

Other Misc. Stuff that I might try to work on (To be updated later):
- [ ] Update the documentation a little bit with some more examples I have laying around 
- [ ] Add Examples for Calibrating a dataset
- [ ] Move some  of the shorter Juptyer Notebooks --> Examples  (Just because the Examples are tested more continually)

tinabe, 2024-05-29:
>       * [ ]  Update ML Notebook @tinabe It might be good to add your notebook for ML to the Pyxem demos!

All good with me. At this point I have to add text explanations, but I will be away from tomorrow including June 4th.. Should this notebook replace any existing demo notebook? As I understand it, the segmentation code used for MgO nanocrystal segmentation will be removed in future versions? If so, would it be a good idea to replace 06 with the Ag example and also remove the machine learning in 01 to make the demos more clearly structured?

@grigoryn94 might also be interested in this post.

CSSFrancis, 2024-05-29:
> All good with me. At this point I have to add text explanations, but I will be away from tomorrow including June 4th.. Should this notebook replace any existing demo notebook? As I understand it, [...]

My plan was to create a "ML" folder to help organize the growing number of tutorials.  The MgO segmentation will stay at least in some capcity, but I think that we should think about how to make it more general :). There are lots of new tools with sklearn, skimage etc that are continually evoloving and I think the idea is that it is fairly difficult to keep up with external tools adding new features to pyxem to keep up.  That's kind of my goal with things like https://pyxem.readthedocs.io/en/stable/reference/generated/pyxem.signals.DiffractionVectors.cluster.html#pyxem.signals.DiffractionVectors.cluster which uses any of the methods avilable in https://scikit-learn.org/stable/modules/clustering.html or with the `filter` method.   Hopefully, then we can focus more on adding new tutorials and using pyxem in a more modular way rather in a strictly linear workflow like fashion.  

> @grigoryn94 might also be interested in this post.

Thanks for pinging @ grigoryn94 as well :)

CSSFrancis, 2024-06-03:
Is there anyone who can help with the review process a little bit over the next couple of days?  It would be nice to merge some of the outstanding PR's and make a rc in the next couple of days so we can make a full release 0.19.0 release before the weekend.

Right now we have:

- [ ] #1076 Which if @viljarjf or @din14970 want to sign off on that would be fantastic!
- [ ] #1081 This one is pretty routine but if @tinabe or @grigoryn94 have any more information they want added about the datasets that might be useful.
- [ ] #1085 This one depends on #1076 getting merged but it should be fairly self explanitory as well.  I'd be interested to know if this works with different calibration datasets etc. 
- [ ] #1082 This is another simple example that could probably be merged.  It might be nice to have someone look over the spelling/ make sure things make sense or suggest things that might make for a better explanation! 
- [ ] #1057 This one is probably squarly on @magnunor and I. If other people working on DPC/magnentic characterization have suggestions and can help review I'd love to hear them.  I'm going to make some small adjustements today (get the tests passing and add simple examples) and see what I can do to clean up the documentation. 

@pc494 @ericpre @hakonanes I realize you all are quite busy at the moment but I wouldn't say no if any of you have any time to review or offer some suggestions.  

I think I owe all of you beer/coffee or both next time I see you :)

hakonanes, 2024-06-03:
I have time on Wednesday and the coming weekend. 🍺 sounds good!

pc494, 2024-06-22:
close as the `rc` went out?

closed 2024-08-07 by pc494