pyxem/pyxem #1078: average_neighbour_patterns

tinabe, 2024-05-15:
Kikuchipy has a handy function for EBSD signals, `average_neighbour_patterns`, with which you can average adjacent patterns weighted by a user-selected window. This can be useful as a pre-processing step for 4D-STEM data as well. As of now, one can define the signal as ebsd and average it using kikuchipy, before changing it back to electrondiffraction2d or similar and continuing the processing using pyxem. However, it would be smoother if it could be done directly in pyxem. I am not aware of this option in pyxem, and wanted to hear your thoughts on this. What do you think @hakonanes?

CSSFrancis, 2024-05-15:
Hi @tinabe I __love__ doing this actually :).  Currently you can do this using the 'filter' function.

https://pyxem.readthedocs.io/en/stable/examples/processing/filtering_data.html#sphx-glr-examples-processing-filtering-data-py

We could think about doing this using more specific functions as well!

tinabe, 2024-05-15:
> Hi @tinabe I **love** doing this actually :). Currently you can do this using the 'filter' function.
> 
> https://pyxem.readthedocs.io/en/stable/examples/processing/filtering_data.html#sphx-glr-examples-processing-filtering-data-py
> 
> We could think about doing this using more specific functions as well!

This makes sense - thank you so much for the quick reply! Then I suppose 'average_neighbour_patterns' should be equivalent to doing the filtering only in real space for diffraction2d signals. Perhaps a sentence could be added to the documentation to make it clearer how the filtering is applied across the signal and navigation axes and include the words averaging of neighbouring patterns to make it appear if one searches for that?

closed 2024-06-19 by tinabe