pyxem/pyxem #1086: Examples for Doing Rotation Correction

CSSFrancis, 2024-06-03:
**Is your feature request related to a problem? Please describe.**
Is there a good way to do Rotation Correction: Scan Rotation --> Detector? This isn't something I ever care about but I assume when doing good orientation mapping it matters?  

It would be great to add an example of how to do this correction (and how to apply an affine transform?) To handle this when doing orientation mapping.

pc494, 2024-06-22:
I think the short answer is "No, there isn't a good way". Standard practice was to do everything in k-space and right at the very end apply a single correction to the orientation map, although I only ever cared about misorientations so it was never an issue for me.

CSSFrancis, 2024-06-22:
@pc494 After talking with @tinabe and @viljarjf it seems like this is best done at the end. Although with the new azimuthal integration you can just redefine the start and stop angles.

It might be good to have this saved in the metadata under the calibration object and then just automatically apply it if is there.